armbian/build #1447: Add stage after kernel patching to apply customizations prior to compile stage

lanefu, 2019-07-04:
Currently unique workarounds are embedded directly into `compilation.sh`   This functionality needs to be moved out of the core script, and loaded and applied dynamically, similar to our patching process. 

Original Request:

Igor: "Expand patching to run scripts/recipes the same way as patches. After all patches are applied, right before compilation starts. Purpose is to solve some existing exceptions and make a proper way of handling new ones."

As such:
https://github.com/armbian/build/blob/master/lib/compilation-prepare.sh

https://github.com/armbian/build/blob/master/lib/compilation.sh#L168-L179

g-provost, 2019-07-08:
So this Issues doesn't apply only to Kernel but to U-boot also ?

Might be fastidious but any change that you guys list all the exceptions that need to be fixed and create an issue for each and assigned it to someone.

igorpecovnik, 2019-07-08:
> So this Issues doesn't apply only to Kernel but to U-boot also ?

This enhancement shell follow the patching logic and the idea is to make a structure for running additional scripts before the (atf, u-boot, kernel) compilation begins. This will help in a way to cleanup files which @lanefu exposed and also from config/sources  

If we agree that this is a good way & if you/anyone understand what has to be done, pick the issue up.

lanefu, 2019-07-08:
> So this Issues doesn't apply only to Kernel but to U-boot also ?
> 
> Might be fastidious but any change that you guys list all the exceptions that need to be fixed and create an issue for each and assigned it to someone.

Your approach is sound, but For now lets avoid creating what i would consiser subtasks for now and track them in the issue.   We can adjust as neededed of course.

closed 2021-06-14 by lanefu