armbian/build #1457: Debian Buster reports suite change on apt update

lanefu, 2019-07-13:
```
root@orangepiplus2e:~/download# apt update
Get:2 http://security.debian.org buster/updates InRelease [39.1 kB]
Get:1 http://cdn-fastly.deb.debian.org/debian buster InRelease [118 kB]
Get:4 http://cdn-fastly.deb.debian.org/debian buster-updates InRelease [46.8 kB]
Hit:5 http://cdn-fastly.deb.debian.org/debian buster-backports InRelease        
Hit:3 https://beta.armbian.com buster InRelease    
N: Repository 'http://cdn-fastly.deb.debian.org/debian buster InRelease' changed its 'Version' value from '' to '10.0'
E: Repository 'http://cdn-fastly.deb.debian.org/debian buster InRelease' changed its 'Suite' value from 'testing' to 'stable'
N: This must be accepted explicitly before updates for this repository can be applied. See apt-secure(8) manpage for details.
Do you want to accept these changes and continue updating from this repository? [y/N] y  
Get:6 http://cdn-fastly.deb.debian.org/debian buster/main armhf Packages [7688 kB]
N: Repository 'http://security.debian.org buster/updates InRelease' changed its 'Version' value from '' to '10'
E: Repository 'http://security.debian.org buster/updates InRelease' changed its 'Suite' value from 'testing' to 'stable'
N: This must be accepted explicitly before updates for this repository can be applied. See apt-secure(8) manpage for details.
Do you want to accept these changes and continue updating from this repository? [y/N] y
Get:7 http://security.debian.org buster/updates/main armhf Packages [11.9 kB]
Get:8 http://cdn-fastly.deb.debian.org/debian buster/main armhf Contents (deb) [35.2 MB]
E: Repository 'http://cdn-fastly.deb.debian.org/debian buster-updates InRelease' changed its 'Suite' value from 'testing-updates' to 'stable-updates'                                                                                                                                    
N: This must be accepted explicitly before updates for this repository can be applied. See apt-secure(8) manpage for details.
Do you want to accept these changes and continue updating from this repository? [y/N] y
Get:9 http://cdn-fastly.deb.debian.org/debian buster/contrib armhf Packages [40.9 kB]                                                                                                                                                                                                    
Get:10 http://cdn-fastly.deb.debian.org/debian buster/contrib armhf Contents (deb) [83.3 kB]                                                                                                                                                                                             
Get:11 http://cdn-fastly.deb.debian.org/debian buster/non-free armhf Contents (deb) [764 kB]                  
```

guysoft, 2019-07-14:
Workaround:
```
apt-get update --allow-releaseinfo-change
```
Related issue in Raspbian:
https://github.com/RPi-Distro/pi-gen/issues/309

lanefu, 2019-07-14:
looks like it was based on our filesystem snapshot as well.... i did a fresh build and its fine.

thanks @guysoft

closed 2019-07-14 by lanefu

igorpecovnik, 2019-07-15:
Cache rebuilding is in progress. after that it should be fine.