armbian/build #1465: Additional kernel modules for mainline Odroid C1

alpe12, 2019-07-20:
Same as https://github.com/armbian/build/issues/113 but for Odroid C1.

```
Could not run iotop as some of the requirements are not met:
- Linux >= 2.6.20 with
  - I/O accounting support (CONFIG_TASKSTATS, CONFIG_TASK_DELAY_ACCT, CONFIG_TASK_IO_ACCOUNTING)
```

igorpecovnik, 2019-07-20:
Please prepare a PR with changes.

lanefu, 2019-07-23:
@alpe12 see https://docs.armbian.com/Process_Contribute/

you'll need to update
 `config/kernel/linux-odroidc1-dev.config`

igorpecovnik, 2019-07-23:
Actually linux-odroidc1-next.config

lanefu, 2019-07-23:
@igorpecovnik 

```
lane@cranbone:~/GIT/ARMBIAN/build/config/kernel$ ls -l *odroidc1*
-rw-r--r-- 1 lane megawheel 115826 Jul 23 02:41 linux-odroidc1-default.config
-rw-r--r-- 1 lane megawheel 188280 Jul 23 02:41 linux-odroidc1-dev.config
```

igorpecovnik, 2019-07-23:
https://github.com/armbian/build/tree/master/config/kernel

git pull ;)

closed 2019-07-23 by alpe12