armbian/build #1472: Mount image fails without error due to use of /loop0p1

Stadicus, 2019-07-22:
For some time now, the build script for Armbian, run in Docker privileged mode, fails at the end when mounting the image on the loopback device. The build script does not fail, it just produces an invalid image.

Strangely, running the build process a second time (always using caching `CLEAN_LEVEL=oldcache`) runs without errors and produces the correct image.

```
[ o.k. ] Preparing image file for rootfs [ rockpro64 bionic ]
[ o.k. ] Current rootfs size [ 1057 MiB ]
[ o.k. ] Creating blank image for rootfs [ 1344 MiB ]
1.31GiB [ 175MiB/s] [===================================================================================================================================================================================================================>] 100%
[ o.k. ] Creating partitions [ root: ext4 ]
[ .... ] Creating rootfs [ ext4 on /dev/loop0p1 ]
mkfs.ext4: No such device or address while trying to determine filesystem size
tune2fs: No such device or address while trying to open /dev/loop0p1
Couldn't find valid filesystem superblock.
mount: /root/armbian/.tmp/mount-default-rockpro64-bionic-no: /dev/loop0p1 is not a valid block device.
[ .... ] Updating initramfs... [ update-initramfs -uv -k 4.4.184-rockchip64 ]
[ o.k. ] Updated initramfs. [ for details see: /root/armbian/output/debug/install.log ]
[ .... ] Copying files to root directory 
        980.78M  99%  258.52MB/s    0:00:03 (xfr#31107, to-chk=0/44341)   

sent 983.46M bytes  received 645.60K bytes  281.17M bytes/sec
total size is 988.04M  speedup is 1.00
[ .... ] Copying files to /boot directory 
         49.01M  99%  392.54MB/s    0:00:00 (xfr#129, to-chk=0/135) 

sent 49.03M bytes  received 2.49K bytes  98.07M bytes/sec
total size is 49.01M  speedup is 1.00
[ o.k. ] Free space: [ SD card ]
Filesystem      Size  Used Avail Use% Mounted on
overlay         201G  164G   27G  87% /
tmpfs            64M     0   64M   0% /dev
tmpfs           7.7G     0  7.7G   0% /sys/fs/cgroup
/dev/nvme0n1p3  201G  164G   27G  87% /root/.ccache
udev            7.7G     0  7.7G   0% /tmp/dev
tmpfs           7.7G   22M  7.7G   1% /tmp/dev/shm
/dev/sda1       1.8T  1.1T  652G  63% /root/armbian
shm              64M     0   64M   0% /dev/shm
tmpfs           7.7G     0  7.7G   0% /proc/asound
tmpfs           7.7G     0  7.7G   0% /proc/acpi
tmpfs           7.7G     0  7.7G   0% /proc/scsi
tmpfs           7.7G     0  7.7G   0% /sys/firmware
tmpfs            11G  1.1G  9.2G  11% /root/armbian/.tmp/rootfs-default-rockpro64-bionic-no
[ o.k. ] Writing U-boot bootloader [ /dev/loop0 ]
umount: /root/armbian/.tmp/mount-default-rockpro64-bionic-no: not mounted.
[ .... ] Fingerprinting 
[ o.k. ] Done building [ /root/armbian/output/images/Armbian_5.91_Rockpro64_Ubuntu_bionic_default_4.4.184.img ]
umount: /root/armbian/.tmp/rootfs-default-rockpro64-bionic-no: target is busy.
rm: cannot remove '/root/armbian/.tmp/rootfs-default-rockpro64-bionic-no': Device or resource busy
[ o.k. ] Runtime [ 20 min ]
[ o.k. ] Repeat Build Options [ ./compile.sh BOARD=rockpro64 BRANCH=default RELEASE=bionic BUILD_DESKTOP=no KERNEL_ONLY=no KERNEL_CONFIGURE=no  ]
```

I looked through the scripts and especially through the following section, but to no avail. 
https://github.com/armbian/build/blob/6dceae89ce35a4f7145003ddbcc9a53ed8999ff8/lib/debootstrap.sh#L416-L453

The host is running Ubuntu 18.04, with the following loopback and other blockdevices present:
````
NAME           MAJ:MIN RM   SIZE RO TYPE  MOUNTPOINT
loop1            7:1    0  54.4M  1 loop  /snap/core18/1055
loop2            7:2    0  88.4M  1 loop  /snap/core/7169
loop3            7:3    0  88.5M  1 loop  /snap/core/7270
loop4            7:4    0 219.1M  1 loop  /snap/gimp/189
loop5            7:5    0 218.9M  1 loop  /snap/gimp/165
loop6            7:6    0  54.4M  1 loop  /snap/core18/1066
loop7            7:7    0   236M  1 loop  /snap/kde-frameworks-5/27
loop8            7:8    0 209.8M  1 loop  /snap/firefox/237
loop9            7:9    0  35.3M  1 loop  /snap/gtk-common-themes/1198
loop10           7:10   0  42.8M  1 loop  /snap/gtk-common-themes/1313
loop11           7:11   0   134M  1 loop  /snap/code/10
loop12           7:12   0   3.3M  1 loop  /snap/konversation/3
loop13           7:13   0 133.9M  1 loop  /snap/code/11
loop14           7:14   0 209.8M  1 loop  /snap/firefox/243
└─loop14p1     259:10   0   1.3G  0 loop  
sda              8:0    0   1.8T  0 disk  
└─sda1           8:1    0   1.8T  0 part  /mnt/hdd
nvme0n1        259:0    0 238.5G  0 disk  
├─nvme0n1p1    259:1    0 238.4M  0 part  /boot/efi
├─nvme0n1p2    259:2    0   3.5G  0 part  
├─nvme0n1p3    259:3    0 203.5G  0 part  /
└─nvme0n1p4    259:4    0  31.3G  0 part  
  └─cryptswap1 253:0    0  31.3G  0 crypt [SWAP]
````

Having the build process fail after 20min is quite painful. Anything special to consider? Any additional information I can provide?

lanefu, 2019-07-23:
@Stadicus 

are you building with `./compile.sh docker`?  That's the only supported method per https://docs.armbian.com/Developer-Guide_Building-with-Docker/

if you look in the build scripts you'll see that its explicitly running mknod for all the /dev/loopXX devices so that they exist in the container.

igorpecovnik, 2019-07-23:
Try to revert back this commit: https://github.com/armbian/build/commit/aea890200c0c2ab8156a31eb991a087ab54d8ec0#diff-cc2879263e4a73e37cf60f2fa6b3fb64

lanefu, 2019-07-27:
@Stadicus 

Any updates?

igorpecovnik, 2019-07-27:
It happened to me, once. Could not catch up anything yet. Debian Buster host.

Stadicus, 2019-07-27:
Did try to build using the commit mentioned, but am not quite sure how to do that. Even when checking out the specific tag, it switches back to master branch. Can I specify it as TAG in the build options?

Currently away for a week, will take it up right after again.

lanefu, 2019-07-27:
`touch .ignore_changes` should keep it from checking out master 

Ill try to add that to the documentation this weekend

Stadicus, 2019-08-02:
Thanks for the hints, building on this commit https://github.com/armbian/build/commit/aea890200c0c2ab8156a31eb991a087ab54d8ec0#diff-cc2879263e4a73e37cf60f2fa6b3fb64, together with the `.ignore_changes` setting, works.

Built the image four times succesfully, using various configuration, with and without cache.

How would I proceed from here, to not be stuck on this old commit?

```
[ o.k. ] Preparing image file for rootfs [ rockpro64 bionic ]
[ o.k. ] Current rootfs size [ 1057 MiB ]
[ o.k. ] Creating blank image for rootfs [ 1344 MiB ]
[ o.k. ] Creating partitions [ root: ext4 ]
[ .... ] Creating device node [ /dev/loop14p1 ]
[ .... ] Creating rootfs [ ext4 on /dev/loop14p1 ]
[ .... ] Updating initramfs... [ update-initramfs -uv -k 4.4.178-rockchip64 ]
[ o.k. ] Updated initramfs. [ for details see: /root/armbian/output/debug/install.log ]
[ .... ] Copying files to root directory 
        980.98M  99%   55.07MB/s    0:00:16 (xfr#31113, to-chk=0/44347)   

sent 983.65M bytes  received 645.72K bytes  56.25M bytes/sec
total size is 988.24M  speedup is 1.00
[ .... ] Copying files to /boot directory 
         62.88M  99%   28.29MB/s    0:00:02 (xfr#131, to-chk=0/138) 

sent 62.90M bytes  received 2.53K bytes  25.16M bytes/sec
total size is 62.88M  speedup is 1.00
[ o.k. ] Free space: [ SD card ]
Filesystem      Size  Used Avail Use% Mounted on
overlay         201G  166G   25G  87% /
tmpfs            64M     0   64M   0% /dev
tmpfs           7.7G     0  7.7G   0% /sys/fs/cgroup
shm              64M     0   64M   0% /dev/shm
udev            7.7G     0  7.7G   0% /tmp/dev
tmpfs           7.7G  128M  7.6G   2% /tmp/dev/shm
/dev/sda1       1.8T  1.1T  639G  64% /root/armbian
/dev/nvme0n1p3  201G  166G   25G  87% /root/.ccache
tmpfs           7.7G     0  7.7G   0% /proc/asound
tmpfs           7.7G     0  7.7G   0% /proc/acpi
tmpfs           7.7G     0  7.7G   0% /proc/scsi
tmpfs           7.7G     0  7.7G   0% /sys/firmware
tmpfs            11G  1.1G  9.2G  11% /root/armbian/.tmp/rootfs-default-rockpro64-bionic-no
/dev/loop14p1   1.3G  1.1G  146M  89% /root/armbian/.tmp/mount-default-rockpro64-bionic-no
[ o.k. ] Writing U-boot bootloader [ /dev/loop14 ]
[ o.k. ] Done building [ /root/armbian/output/images/Armbian_5.82_Rockpro64_Ubuntu_bionic_default_4.4.178.img ]
umount: /root/armbian/.tmp/rootfs-default-rockpro64-bionic-no: target is busy.
rm: cannot remove '/root/armbian/.tmp/rootfs-default-rockpro64-bionic-no': Device or resource busy
[ o.k. ] Runtime [ 40 min ]
[ o.k. ] Repeat Build Options [ BOARD=rockpro64 BRANCH=default RELEASE=bionic BUILD_DESKTOP=no KERNEL_ONLY=no KERNEL_CONFIGURE=no ]
```

Stadicus, 2019-08-07:
How would I proceed from here, to not be stuck on this old commit?

Stadicus, 2019-08-21:
Not sure what changed, either in this codebase or on my machine locally, but I have not experienced this issue within the last 1-2 weeks, whereas it was easy to reproduce every time beforehand.

igorpecovnik, 2019-08-21:
OK, tnx, closing.

closed 2019-08-21 by igorpecovnik