armbian/build #1474: remove merged patches!!

zhangn1985, 2019-07-23:
due to lima is merged to kernel 5.2.y, this means lima patches for linux-5.2.y dev are not needed, but while compiling orangepipc dev, all lima patches are applied seccusfully. 

why?

because, lima patches add lots of files, and these files will be removed before apply patches.

please check this file:
https://github.com/armbian/build/blob/master/lib/compilation.sh#L588

please remove this line, and make sure all files created by patches are remove during git repo checkout and clean.

igorpecovnik, 2019-07-23:
> due to lima is merged to kernel 5.2.y, this means lima patches for linux-5.2.y dev are not needed, but while compiling orangepipc dev, all lima patches are applied seccusfully.
why?

Because someone needs to check which one are deprecated and which not. This is development branch, so no need to rush anywhere. Usability of Lima is virtually none at this moment. Too slow.

> please remove this line, and make sure all files created by patches are remove during git repo checkout and clean.

This line is just fine, but perhaps we need to add some verbosity.

zhangn1985, 2019-07-23:
no, lima is just a break point I look into this issue. why merged patches are still applied seccusfully.

for sunxi-dev:
uboot,
[ warn ] * [l][c] add-nanopineoplus2.patch [ failed ]
[ warn ] * [l][c] add-orangepi-zeroplus.patch [ failed 

kernel, exclude lima,
[ warn ] * [l][c] 0117-ASoC-dt-binding-Add-bindings-for-Allwinner-A64-codec.patch [ failed ]
[ warn ] * [l][c] 0134-drm-sun4i-rgb-Add-5-tolerance-to-dot-clock-frequency.patch [ failed ]
[ warn ] * [l][c] board-a64-v3-02-19-arm64-dts-allwinner-a64-Add-Pine64-LTS-device-tree-file.patch [ failed ]
[ warn ] * [l][c] general-armbian-boot-logo-on-screen-center.patch [ failed ]

re-apply merged patch means apply **out-of-date patches**, you just lucky that no compile failure.
or function issue.

igorpecovnik, 2019-07-23:
When we go up with a kernel version, some patches become obsolete, some need adjustments. In this process we learn which are not needed. Not before.

> re-apply merged patch means apply out-of-date patches

You don't know if they are out-of-date. They are probably the same ... but I agree that some RFC is needed here. Not by removing that line, but adding one IF CREATE_PATCHES="yes" there. (I think that's why we have this cleanup there, but I am not 100% sure. Can't dig deeper now)

> you just lucky that no compile failure.

Most likely over patches only overwrite with the same content. 

> for sunxi-dev:

Look into the logs why? Perhaps those patches also become obsolete ... but its always good to check the file which is overwritten. Sometimes (mostly) upstream is behind ours.

igorpecovnik, 2019-07-23:
Why cleaning up: https://forum.armbian.com/topic/7885-incomplete-clean-up-of-source-code/

zhangn1985, 2019-07-23:
https://github.com/armbian/build/commit/eb528f6c9e550bdadf4b8fe742985340041fbfac
this patch is enough to cleaning up, no need this line:
https://github.com/armbian/build/blob/master/lib/compilation.sh#L588

and again this is a chance to maintain less patches, do you agree?

zhangn1985, 2019-07-23:
`git help clean`

       -d
           Remove untracked directories in addition to untracked files. If an untracked directory is managed by a different Git repository, it is not removed by default. Use -f option twice if you really
           want to remove such a directory.
       -f, --force
           If the Git configuration variable clean.requireForce is not set to false, git clean will refuse to delete files or directories unless given -f, -n or -i. Git will refuse to delete directories
           with .git sub directory or file unless a second -f is given
       -q, --quiet
           Be quiet, only report errors, but not the files that are successfully removed.

igorpecovnik, 2019-07-23:
> chance to maintain less patches, do you agree?

Yes, we need to remove those who are pointless. We do this job for several years now and method how we do that is improving all the time. But it can't be perfect from the day one.

Core functionality will not be changed without deep investigation which I can't afford on a short notice. I am out on vacations until end of the months and things related to DEV are not something I will wake up in the middle of the night ;)

zhangn1985, 2019-07-23:
> But it can't be perfect from the day one.

this requires board/family maintainer work togother , not one man's job.

how about create a cleanup branch?

> Nope, we don't want that.

No, you already did that.

zhangn1985, 2019-07-23:
in https://github.com/armbian/build/commit/eb528f6c9e550bdadf4b8fe742985340041fbfac , it removes ```Files that are not tracked by git and were added```, and the files are added by patches.

 https://github.com/armbian/build/blob/master/lib/compilation.sh#L588 it redo it again with another method.

igorpecovnik, 2019-07-23:
The only problem I see here is that there is no verbosity when a patch is creating things that already exists ... that pointless patch can be removed patches. But since there is no rush ... wait,

> how about create a cleanup branch?

Nope. 

All what have to be done is cleaning all Lima patches and close this issue. And open a new one to RFC the patch applying part to get more info ... That's all.

zhangn1985, 2019-07-23:
> in [eb528f6](https://github.com/armbian/build/commit/eb528f6c9e550bdadf4b8fe742985340041fbfac) , it removes `Files that are not tracked by git and were added`, and the files are added by patches.
> 
> https://github.com/armbian/build/blob/master/lib/compilation.sh#L588 it redo it again with another method.

I guess I can't persuade you on remove this line https://github.com/armbian/build/blob/master/lib/compilation.sh#L588, for its danger.

zhangn1985, 2019-07-23:
one more thing about this line https://github.com/armbian/build/blob/master/lib/compilation.sh#L588, if we don't remove this line, we can't find a full set of patches which is merged, has confilct.

zhangn1985, 2019-07-23:
or if you add `git status` after line https://github.com/armbian/build/blob/6dceae89ce35a4f7145003ddbcc9a53ed8999ff8/lib/general.sh#L272

you will get `nothing to commit, working tree clean`

that means you already get a clean source, not need to clean it again by: https://github.com/armbian/build/blob/master/lib/compilation.sh#L588

igorpecovnik, 2019-07-23:
> if we don't remove this line, we can't find a full set of patches which is merged, has confilct.

I know that. I propose expanding that line to tell us what it is deleting and why. Not just removing. I am trying to understand if this has any effects with CREATE_PATCHES ... but am **too detached** to approve core changes and afraid that I will need to dig deep because of possible problems while ... still vacations. Other will need to approve, if.

zhangn1985, 2019-07-23:
get your idea. let me check CREATE_PATCHES together with you.

zhangn1985, 2019-07-23:
I think there is no impact to CREATE_PATCHES, because CREATE_PATCHES is done after applying patches.
and CREATE_PATCHES is in user interactive mode, even there is some issue, user can correct it.

zhangn1985, 2019-07-24:
> scripts/Makefile.build:278: recipe for target 'drivers/gpu/drm/lima/lima_drv.o' failed
> scripts/Makefile.build:489: recipe for target 'drivers/gpu/drm/lima' failed
>   CC [M]  drivers/hwmon/ds620.o
> scripts/Makefile.build:489: recipe for target 'drivers/gpu/drm' failed
> scripts/Makefile.build:489: recipe for target 'drivers/gpu' failed

LIMA is disabled in sunxi-dev, that's why lucky.

lanefu, 2019-07-26:
@zhangn1985 

Can we close issue now?

closed 2019-07-26 by zhangn1985