armbian/build #1481: OrangePi 3 U-Boot configuration produces unusable boot code

sallevan, 2019-07-26:
After applying your patches U-Boot just hangs immediately on SPL stage. Seems that orangepi_3_defconfig created by your patch is invalid -- the following line is missing:

CONFIG_SPL_TEXT_BASE=0x20060

After adding it to configuration U-Boot become working.

igorpecovnik, 2019-07-26:
Are you building with our (unmodified) tools or you do it manually?

sallevan, 2019-07-26:
I build it manually from latest published sources (2019.07).

igorpecovnik, 2019-07-26:
Armbian does not use that sources yet ... so this is not an issue. Please use forum for discussing such problems https://forum.armbian.com/forum/18-allwinner-h6/ This is build script issue tracker. Unrelated to Orange boot problems ...

closed 2019-07-26 by igorpecovnik