armbian/build #1493: Enable bmp280 kernel module on mvebu

count-doku, 2019-08-01:
I'd like to enable the bmp280 kernel modules in the mvebu kernel. Any objections? Otherwise I will upload changes to mvebu branch and file a PR.

Previously done on some other platform: https://github.com/armbian/build/pull/1380

igorpecovnik, 2019-08-01:
No objections. Thinks that are safe, just enable if you find them useful.

aprayoga, 2019-08-02:
i have no objection

g-provost, 2019-08-02:
Yes please go ahead.

-- 
*Gauthier Provost*

*Kobol Innovations Pte. Ltd.*
*101 Cecil Street*
*#26-01/07 Tong Eng Building*
*Singapore 069533*

*e: gauthier@kobol.io <gauthier@kobol.io>*

*m: +65 8118 4507*

*w: https://kobol.io <https://kobol.io>*

* <https://kobol.io/helios4/>*

*UEN 201821321E*

closed 2019-08-02 by count-doku