armbian/build #1497: Cannot compile image for Banana M2U

ruvps, 2019-08-02:
I got this every time

dpkg-deb: building package 'linux-headers-dev-sunxi' in '../linux-headers-dev-sunxi_5.92_armhf.deb'.
dpkg-deb: building package 'linux-dtb-dev-sunxi' in '../linux-dtb-dev-sunxi_5.92_armhf.deb'.
dpkg-deb: building package 'linux-image-dev-sunxi' in '../linux-image-dev-sunxi_5.92_armhf.deb'.
dpkg-buildpackage: info: binary-only upload (no source included)
dpkg-deb: building package 'linux-source-5.2.5-dev-sunxi' in '/tmp/build/.tmp/linux-source-dev-sunxi_5.92_all.deb'.
[ o.k. ] Creating board support package [ bananapim2ultra dev ]
[ .... ] Fingerprinting 
[ o.k. ] Building package [ linux-buster-root-dev-bananapim2ultra ]
[ o.k. ] Building desktop package [ armbian-buster-desktop_5.92_all ]
[ o.k. ] Starting rootfs and image building process for [ bananapim2ultra buster ]
[ o.k. ] Checking for local cache [ buster-cli-armhf.d2f...ce3.tar.lz4 ]
[ .... ] searching on servers 
[ o.k. ] ... remote not found [ Creating new rootfs cache for buster ]
[ o.k. ] Installing base system [ Stage 1/2 ]
mknod: /tmp/build/.tmp/rootfs-dev-bananapim2ultra-buster-no/test-dev-null: Operation not permitted
E: Cannot install into target '/tmp/build/.tmp/rootfs-dev-bananapim2ultra-buster-no' mounted with noexec or nodev
[ error ] ERROR in function create_rootfs_cache [ debootstrap.sh:142 ]
[ error ] Debootstrap base system first stage failed 
[ o.k. ] Process terminated 
[ error ] ERROR in function unmount_on_exit [ image-helpers.sh:62 ]
[ error ] debootstrap-ng was interrupted 
[ o.k. ] Process terminated

igorpecovnik, 2019-08-02:
Can you supply some build logs? What is your build host? Can you try building under Docker?

ruvps, 2019-08-02:
I try on Proxmox via LXC - may be problem here
Now i try to build on native KVM Ubuntu 18.04 (Bionic)

closed 2019-09-23 by igorpecovnik