armbian/build #1501: Rename the RockPi-4b board to Rock Pi 4

piter75, 2019-08-04:
I was thinking about renaming the `RockPi-4b` board (`rockpi-4b` config file) to `Rock Pi 4` (`rock-pi-4` config file).

Both board models work quite well with Armbian and there is a single [rk3399-rock-pi-4.dts](https://github.com/torvalds/linux/blob/master/arch/arm64/boot/dts/rockchip/rk3399-rock-pi-4.dts) device tree file in kernel.

There is one strange issue with eMMC card on model A but I have probably nailed it and will prepare PR for it soon.

With that in mind having "b" in board name as it is right now may be confusing and may raise additional questions in the forums.

I am willing to prepare PR for the name change and its references in patches, file names and so on...
Are there any places outside `armbian/build` that would need to change?

@igorpecovnik @Tonymac32 @martinayotte @chwe17 what do you think?

igorpecovnik, 2019-08-05:
> Are there any places outside armbian/build that would need to change?

Deleting this
https://dl.armbian.com/rockpi-4b/
and rebuilding new set of images.

piter75, 2019-08-05:
One more thing that came to my mind would be that package names will change, e.g.:
https://apt.armbian.com/pool/main/l/linux-u-boot-rockpi-4b-dev/
https://apt.armbian.com/pool/main/l/linux-buster-root-dev-rockpi-4b/

I am still willing to do this but also started to wonder if this change is worth the hassle and potential package update breaking.

Tonymac32, 2019-12-02:
@piter75 is this still a concern?

piter75, 2019-12-02:
Nope. We have separate board files for these models now.

I am closing it.

closed 2019-12-02 by piter75