armbian/build #1505: Refactor board build targets

igorpecovnik, 2019-08-05:
Current *_TARGETS are badly designed. Perhaps having a build_all_config_file.csv for this? This is for internal usage anyway?

Structure:
board, release, (m)minimal|(c)li|(d)esktop, (d)efault|(n)ext|(d)ev, (s)table|(b)eta

g-provost, 2019-08-13:
@igorpecovnik Would it make sense to introduce in this target.conf a column to define if '**debs + img**'  or '**debs only**' needs to be built ?

I'm asking that in reference to Helios4 where we still need for transition to offer _next stretch_ debs packages but we don't need/want anymore to offer the image. And I guess it's a use case that applies to any board.

igorpecovnik, 2019-08-13:
yes, it would make sense. I will add this functionality.

igorpecovnik, 2019-08-13:
done

closed 2019-09-11 by igorpecovnik