armbian/build #1537: rtl8188eu and rtl88XXau drivers broken with preempt_rt patch set

srussvoll, 2019-08-28:
The two kernel module drivers rtl8188eu and rtl88XXau both work perfectly on all tested kernels without the preempt_rt patch set. The rtl8188eu driver has been used reliably with the preempt_rt patches for several months as described in the next section.

Due to connection speed problems I decided to buy a new USB WiFi dongle based on the RTL8811AU chipset. This turned out not to work at all on any of the tested kernels patched with preempt_rt. Hovewer, interestingly, the rtl8188eu driver causes the same kernel oops as the rtl88XXau driver at least from kernel 4.19.59 and up, but not on kernel 4.19.25 and down.

The next section describes the platform and testing with the rtl8188eu driver.
The section after describes testing with the rtl88XXau driver.

---

I am running sunxi-next on a NanoPi Neo Core2 using the preempt_rt patches, configured with PREEMPT_RT_FULL. I have successfully used the 8188eu kernel module driver for the RTL8188EU chipset for several months, running kernel 4.19.25-rt16-sunxi64. The r8188eu driver is blacklisted. The 8188eu module manages to scan for WiFi networks, and connect to them. However, other common operations like `ip link set [device] down` produces a kernel oops that crashes the entire kernel.

Upgrading to 4.19.59-rt24, the 8188eu driver can scan for WiFi networks, but produces a kernel oops on connecting to a network (entire relevant dmesg output is added at the bottom):

> [200.033141] Unable to handle kernel paging request at virtual address ffff00000ce6a48c

Driver version in kernel 4.19.25 and 4.19.13: `v4.1.4_6773.20130222`.
Driver version in kernel 4.19.59: `v5.3.9_28540.20180627`.

My first guess was that the problem might be introduced in a bump after v4.1, so I tested the [v4.1.8 version](https://github.com/lwfinger/rtl8188eu/tree/v4.1.8_9499) with the 4.19.59-rt24 kernel, but the oops still occurs.

---

I tested the 88XXau driver with kernel 4.19.13, 4.19.25 and 4.19.59 without preempt_rt, and operation is stable.
Driver version in kernel 4.19.13: `v5.1.5_19247.20160830`.
Driver version in kernel 4.19.25 and 4.19.59: `v5.2.20.2_28373.20180619`

Testing with the preempt_rt patches, all of them produce the same oops as the one listed in the previous section. The actual address differs, but the two last hexadecimal digits seem to remain the same.

The Armbian build system fetches the 88XXau driver from the [aircrack-ng/rtl8812au repository](https://github.com/aircrack-ng/rtl8812au). I have tried compiling all the v4.x versions on kernel 4.19.25 and 4.19.13 (since this is the two kernels where the 8188eu driver worked), however they do not compile because it tries to use a timer API that does not exist.

Instead, I have tried [v4.3.22-beta](https://github.com/diederikdehaas/rtl8812AU/tree/driver-4.3.22-beta) on all kernel versions mentioned earlier, but all fail with the same oops as mentioned in the previous section. This repository is based on an official driver from Realtek distributed by Edimax, and only contains simple corrections to make it compile on newer kernels.

---

Since I do not know the details of the patches Armbian applies to the kernel, I do not know if the problem lies here, or if the problem really is in the original Realtek driver implementation.

Since both the preempt_rt patches and Armbian are only released for every few kernel releases, I have looked at the commit history and found kernel 4.19.13, 4.19.25 and 4.19.59 to be released both for Armbian and preempt_rt. Since the build system pulls master at the beginning of the build process, I have used the following commands to bring the desired Armbian commit to the front of master:

```
#!/usr/bin/env bash

# 4.19.13: e5b2927
# 4.19.25: 16af554
# 4.19.59: 7ccc58c
commit=[commit hash]

git checkout ${commit}
git checkout -b old_commit
git merge -s ours master
git checkout master
git merge old_commit
git branch -d old_commit
```

---

Relevant output from dmesg is pasted below. All oopses have the same cause and similar output, and they all happen within the `rt_spin_lock` call.

```
[  200.033141] Unable to handle kernel paging request at virtual address ffff00000ce6a48c
[  200.041062] Mem abort info:
[  200.043853]   ESR = 0x96000021
[  200.046904]   Exception class = DABT (current EL), IL = 32 bits
[  200.052817]   SET = 0, FnV = 0
[  200.055867]   EA = 0, S1PTW = 0
[  200.058994] Data abort info:
[  200.061871]   ISV = 0, ISS = 0x00000021
[  200.065701]   CM = 0, WnR = 0
[  200.068669] swapper pgtable: 4k pages, 48-bit VAs, pgdp = 0000000092c5c328
[  200.075536] [ffff00000ce6a48c] pgd=0000000077ffe803, pud=0000000077ffd803, pmd=0000000069972003, pte=00e800006ef7af13
[... 80 lines cut ...]
[  200.575678]  OnAssocRsp+0x120/0x2b8 [8188eu]
[  200.575730]  _mgt_dispatcher+0x84/0xdc [8188eu]
[  200.575782]  mgt_dispatcher+0x128/0x220 [8188eu]
[  200.575841]  validate_recv_mgnt_frame+0x90/0x164 [8188eu]
[  200.575899]  validate_recv_frame+0x3a0/0x404 [8188eu]
[  200.575957]  recv_func_prehandle+0x40/0x84 [8188eu]
[  200.576014]  recv_func+0x30/0x194 [8188eu]
[  200.576072]  rtw_recv_entry+0x1c/0x54 [8188eu]
[  200.576130]  pre_recv_entry+0x38/0x80 [8188eu]
[  200.576190]  recvbuf2recvframe+0x2b8/0x2ec [8188eu]
[  200.576252]  usb_recv_tasklet+0x70/0x140 [8188eu]
[  200.576256]  tasklet_action_common.isra.2+0xc8/0x1b8
[  200.576259]  tasklet_action+0x20/0x28
[  200.576262]  do_current_softirqs+0x19c/0x230
[  200.576265]  __local_bh_enable+0x54/0x68
[  200.576269]  irq_forced_thread_fn+0x84/0xb0
[  200.576273]  irq_thread+0x12c/0x1d8
[  200.576276]  kthread+0x128/0x130
[  200.576280]  ret_from_fork+0x10/0x1c
[  200.576281] handlers:
[  200.576286] [<0000000019bdfadb>] irq_default_primary_handler threaded [<000000001c57de0c>] usb_hcd_irq
[  200.576295] Disabling IRQ #14
[  200.576531] NOHZ: local_softirq_pending 40
[  200.578789] NOHZ: local_softirq_pending 40
```

igorpecovnik, 2019-08-28:
How we patch the whole thing:
1. We clone sources. Mainline in this case ... [since few weeks](https://github.com/armbian/build/pull/1507), before we were attached to a private fork which only adds complexity
2. [We apply wifi / aufs patches](https://github.com/armbian/build/blob/master/lib/compilation-prepare.sh) 
3. Then [other patches](https://github.com/armbian/build/tree/master/patch/kernel/sunxi-next) gets applied

... and then a kernel gets compiled, packed and installed.

I would prefer if we move this conversation to forum since this issue tracker is not exactly the right place. It's here to collect build script issue. This is kernel or driver which we can try to observe [here](https://forum.armbian.com/forum/4-development/), but I can't promise to dig in deep into this anytime soon. If at all.

srussvoll, 2019-08-29:
Today I have looked at the list of patches being applied to the kernel. At first glance none of them seem to modify the way USB works in the kernel, and none of them patch anything related to the 8188eu and 88XXau drivers. They do add DTB definitions, but considering the drivers work without the preempt_rt patches, this should not be the cause of the problem.

I also had a look at the second step you listed, and it only adds kernel drivers. This should not affect the results I got when compiling and testing the different drivers mentioned in the original issue post.

This leads me to think the issue is likely related to the original Realtek drivers, so I have emailed them to let them know. I will close this issue, as you suggested, since this is not the right place for this problem.

closed 2019-08-29 by srussvoll