armbian/build #1538: armbianEnv.txt missing from odroidxu4 image

Timple, 2019-08-30:
When doing a fresh build the armbianEnv.txt is missing from odroidxu4 image.

However, when I enable a customize-image.sh file in the userpatches folder with the following content:
```bash
#!/bin/bash
echo "Showing contents of /boot/armbianEnv.txt: "
cat /boot/armbianEnv.txt
```
Then I see this being echod in the terminal:
```txt
Showing contents of /boot/armbianEnv.txt: 
verbosity=1
console=both
```
Which is in line with: [odroidxu4-default.txt](https://github.com/armbian/build/blob/master/config/bootenv/odroidxu4-default.txt)

The resulting image however is missing this file, other files seem present in the /boot folder

Any thoughts?

igorpecovnik, 2019-08-30:
No idea from my head. We need to investigate.

Timple, 2019-08-30:
Thanks for the swift reply. 

I just verified the file is also missing from the downloadable images for odroid.

igorpecovnik, 2019-08-30:
What if you build a default image?

Timple, 2019-08-30:
What would be a 'default image'? With desktop?
Or without the customize-image.sh?

Because the default image on your server is also affected.

I'm happy to run the default once I get the parameters.

Current build options:
```
Build Options [ ./compile.sh docker BOARD=odroidxu4 BRANCH=next RELEASE=bionic BUILD_MINIMAL=no BUILD_DESKTOP=no KERNEL_ONLY=no KERNEL_CONFIGURE=no  ]
```

igorpecovnik, 2019-08-30:
Change BRANCH=next -> BRANCH=default and see what happens.

Timple, 2019-08-30:
```
[ o.k. ] Repeat Build Options [ ./compile.sh docker BOARD=odroidxu4 BRANCH=default RELEASE=bionic BUILD_MINIMAL=no BUILD_DESKTOP=no KERNEL_ONLY=no KERNEL_CONFIGURE=no  ]
```
Still no `armbianEnv.txt` file...

igorpecovnik, 2019-08-30:
OK, later ... now can't work on this.

Timple, 2019-08-30:
Alright, let me know if I can check anything more. 
But for now I can trace the file on the filesystem until it's written on the img. And that part is magic for me unfortunately.

igorpecovnik, 2019-08-31:
It looks like its normal. https://github.com/armbian/build/blob/buildall/lib/debootstrap.sh#L490 until we don't switch to mainline u-boot where there is no boot.ini 

And my XU4 is not working anymore ... just strange sounds when powering it.

Timple, 2019-09-03:
We modify the armbianEnv.txt in our customize-image.sh file.

I guess we could make it work with boot.ini as well, but the first lines there state:
```
ODROIDXU-UBOOT-CONFIG

# DO NOT EDIT THIS FILE
#
# Please edit /boot/armbianEnv.txt to set supported parameters
#
```

So is there any preferred way of doing things here?

igorpecovnik, 2019-09-03:
Create this file and check if it loads properly from boot.ini - IMO it should.

There was an idea to move to mainline u-boot, where we have some unification with others, but it was not executed yet ... preasumably for some issues. I wasn't working on this.

Timple, 2019-09-03:
We already do create this file. But afterwards it gets deleted.

We create it at customize-image.sh because we need the `extraargs=net.ifnames=0` in order to create network aliases. (we need dhcp and fixed ips).

Is there an other option to set this at image-generation? Or should we go agains the warning and modify the boot.ini? (Gets overwritten later I'm affraid).

igorpecovnik, 2019-09-03:
Ahaa. Then this is bug after all :) I'll check when possible.

igorpecovnik, 2019-12-12:
Apparently this is not a bug but feature. See: https://github.com/armbian/build/blob/master/lib/debootstrap.sh#L491-L501
Until we will not switch to mainline u-boot. We did once, but there were issues. Closing.

closed 2019-12-12 by igorpecovnik