armbian/build #1540: Build error 'E: unrecognized or invalid option --components='

g40, 2019-09-01:
Attempting to build from a fresh clone of this repo. 

```$ uname -a
Linux ubuntu-1804 4.15.0-30-generic #32-Ubuntu SMP Thu Jul 26 17:42:43 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

$ git status
On branch master
Your branch is up to date with 'origin/master'.

./compile.sh BOARD=nanopik1plus BRANCH=next KERNEL_ONLY=no KERNEL_CONFIGURE=yes RELEASE=Stretch BUILD_DESKTOP=no BUILD_MINIMAL=yes

E: unrecognized or invalid option --components=
[ error ] ERROR in function create_rootfs_cache [ debootstrap.sh:144 ]
[ error ] Debootstrap base system first stage failed 
[ o.k. ] Process terminated 
[ error ] ERROR in function unmount_on_exit [ image-helpers.sh:62 ]
[ error ] debootstrap-ng was interrupted 
[ o.k. ] Process terminated 
```

igorpecovnik, 2019-09-01:
RELEASE=Stretch -> RELEASE=stretch

closed 2019-09-01 by igorpecovnik

g40, 2019-09-01:
Thanks for spotting that.