armbian/build #1550: Orange Pi Win Plus, mainline kernel 5.3-rc8 panic

MaxPotapov, 2019-09-11:
The same thing was observed at 5.1 and 5.2 kernel
Let's eliminate bug.

U-Boot SPL 2019.04-armbian (Sep 09 2019 - 23:42:00 +0300)
DRAM: 2048 MiB
Trying to boot from MMC1
NOTICE:  BL3-1: Running on A64/H64 (1689) in SRAM A2 (@0x44000)
NOTICE:  Configuring SPC Controller
NOTICE:  BL3-1: v1.0(debug):c9f55c0
NOTICE:  BL3-1: Built : 23:41:56, Sep  9 2019
NOTICE:  DT: sun50i-a64-orangepi-win
INFO:    Configuring AXP PMIC
INFO:    PMIC: DRAM voltage: 1.50V
INFO:    PMIC: setup successful
NOTICE:  SCPI: dummy stub handler, implementation level: 000000
INFO:    BL3-1: Initializing runtime services
INFO:    BL3-1: Preparing for EL3 exit to normal world
INFO:    BL3-1: Next image address: 0x4a000000, SPSR: 0x3c9

U-Boot 2019.04-armbian (Sep 09 2019 - 23:42:00 +0300) Allwinner Technology

CPU:   Allwinner A64 (SUN50I)
Model: OrangePi Win/Win Plus
DRAM:  2 GiB
MMC:   Device 'mmc@1c11000': seq 1 is in use by 'mmc@1c10000'
mmc@1c0f000: 0, mmc@1c10000: 2, mmc@1c11000: 1
Loading Environment from EXT4... ** File not found /boot/boot.env **

** Unable to read "/boot/boot.env" from mmc0:1 **
In:    serial
Out:   serial
Err:   serial
Net:   phy interface7
eth0: ethernet@1c30000
230454 bytes read in 13 ms (16.9 MiB/s)
starting USB...
USB0:   USB EHCI 1.00
USB1:   USB OHCI 1.0
USB2:   USB EHCI 1.00
USB3:   USB OHCI 1.0
scanning bus 0 for devices... 1 USB Device(s) found
scanning bus 1 for devices... 1 USB Device(s) found
scanning bus 2 for devices... 1 USB Device(s) found
scanning bus 3 for devices... 1 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found
Autoboot in 1 seconds, press <Space> to stop
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot/boot.scr
3042 bytes read in 2 ms (1.5 MiB/s)
## Executing script at 4fc00000
U-boot loaded from SD
Boot script loaded from mmc
153 bytes read in 1 ms (149.4 KiB/s)
37246 bytes read in 6 ms (5.9 MiB/s)
3801 bytes read in 3 ms (1.2 MiB/s)
Applying kernel provided DT fixup script (sun50i-a64-fixup.scr)
## Executing script at 44000000
9136698 bytes read in 449 ms (19.4 MiB/s)
15147016 bytes read in 742 ms (19.5 MiB/s)
## Loading init Ramdisk from Legacy Image at 4fe00000 ...
   Image Name:   uInitrd
   Image Type:   AArch64 Linux RAMDisk Image (gzip compressed)
   Data Size:    9136634 Bytes = 8.7 MiB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 4fa00000
   Booting using the fdt blob at 0x4fa00000
EHCI failed to shut down host controller.
   Loading Ramdisk to 49749000, end 49fff9fa ... OK
   Loading Device Tree to 00000000496d7000, end 0000000049748fff ... OK

Starting kernel ...

[   12.224770] Unable to handle kernel paging request at virtual address ffff000014136030
[   12.224774] Unable to handle kernel NULL pointer dereference at virtual address 0000000000000080
[   12.224777] ------------[ cut here ]------------
[   12.224784] Mem abort info:
[   12.224787] kernel BUG at arch/arm64/kernel/traps.c:403!
[   12.224792] Internal error: Oops - BUG: 0 [#1] SMP
[   12.224796] Modules linked in: zstd zram brcmfmac brcmutil snd_soc_hdmi_codec cfg80211 rfkill snd_soc_simple_card snd_soc_spdif_tx snd_soc_simple_card_utils snd_soc_core snd_pcm_dmaengine snd_pcm axp20x_battery axp20x_ac_power snd_timer axp20x_usb_power axp20x_adc lima dw_hdmi_cec dw_hdmi_i2s_audio snd soundcore sun4i_gpadc_iio gpu_sched industrialio cpufreq_dt pinctrl_axp209 realtek dwmac_sun8i mdio_mux
[   12.224849] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 5.3.0-rc8-sunxi64 #5.96
[   12.224851] Hardware name: OrangePi Win/Win Plus (DT)
[   12.224855] pstate: 00000005 (nzcv daif -PAN -UAO)
[   12.224869] pc : do_undefinstr+0x29c/0x2b8
[   12.224872] lr : do_undefinstr+0x138/0x2b8
[   12.224874] sp : ffff000010003ca0
[   12.224876] x29: ffff000010003ca0 x28: ffff000010e50b40
[   12.224881] x27: 0000000000000020 x26: 0000000000000000
[   12.224885] x25: ffff000010e63000 x24: 0000000000000000
[   12.224889] x23: 0000000020000005 x22: ffff0000101200a4
[   12.224893] x21: ffff000010003e40 x20: ffff000010003d00
[   12.224897] x19: ffff000010e48000 x18: 0000000000000000
[   12.224901] x17: 0000000000000000 x16: 0000000000000000
[   12.224905] x15: 0000000000000000 x14: 0000000000000000
[   12.224909] x13: 0000000000000000 x12: 0000000000000000
[   12.224912] x11: 0000000000000001 x10: ffff7e0001ccac60
[   12.224916] x9 : 0000000000000000 x8 : 0000000000000060
[   12.224920] x7 : ffff000010e486c8 x6 : ffff000010003cf8
[   12.224924] x5 : ffff000010ef70f8 x4 : 0000000000000000
[   12.224928] x3 : 00000000d5300000 x2 : ffff000010e52660
[   12.224932] x1 : ffff000010ef70f8 x0 : 0000000020000005
[   12.224936] Call trace:
[   12.224941]  do_undefinstr+0x29c/0x2b8
[   12.224946]  el1_undef+0x10/0x84
[   12.224953]  rcu_core+0x20c/0x470
[   12.224957]  rcu_core_si+0xc/0x18
[   12.224961]  __do_softirq+0x114/0x22c
[   12.224966]  irq_exit+0xac/0xc0
[   12.224973]  __handle_domain_irq+0x60/0xb0
[   12.224976]  gic_handle_irq+0x58/0xa8
[   12.224979]  el1_irq+0xb8/0x140
[   12.224985]  arch_cpu_idle+0x10/0x18
[   12.224991]  do_idle+0x1e0/0x2c0
[   12.224996]  cpu_startup_entry+0x24/0x28
[   12.225004]  rest_init+0xb4/0xc0
[   12.225009]  arch_call_rest_init+0xc/0x14
[   12.225013]  start_kernel+0x44c/0x478
[   12.225020] Code: 2a154035 17ffffae a9025bb5 f9001bb7 (d4210000)
[   12.225025] ---[ end trace 95b82daffffb037f ]---
[   12.225029] Kernel panic - not syncing: Fatal exception in interrupt
[   12.225035] SMP: stopping secondary CPUs
[   12.232689] Mem abort info:
[   12.241455]   ESR = 0x96000004
[   12.246060]   ESR = 0x86000006
[   12.248846]   Exception class = DABT (current EL), IL = 32 bits
[   12.254144]   Exception class = IABT (current EL), IL = 32 bits
[   12.258923]   SET = 0, FnV = 0
[   12.294642]   SET = 0, FnV = 0
[   12.301761]   EA = 0, S1PTW = 0
[   12.306799]   EA = 0, S1PTW = 0
[   12.311576] Data abort info:
[   12.315662] swapper pgtable: 4k pages, 48-bit VAs, pgdp=0000000040d73000
[   12.319745]   ISV = 0, ISS = 0x00000004
[   12.323050] [ffff000014136030] pgd=00000000b7fff003, pud=00000000b7ffe003, pmd=0000000000000000
[   12.328348]   CM = 0, WnR = 0
[   12.536707] user pgtable: 4k pages, 48-bit VAs, pgdp=00000000b35ed000
[   12.543134] [0000000000000080] pgd=0000000000000000
[   13.321519] SMP: failed to stop secondary CPUs 0-2
[   13.326301] Kernel Offset: disabled
[   13.329783] CPU features: 0x0002,24002004
[   13.333784] Memory Limit: none
[   13.336835] Rebooting in 10 seconds..

martinayotte, 2019-09-11:
That is  bit strange since my OPIWin was running fine with 5.0.y, 5.1.y, 5.2.y and now with 5.3.0-rc3 .
How do you power your board ?
Are you using barrel jack or MicroUSB ?
If it is by MicroUSB, the problem is probably there ...

MaxPotapov, 2019-09-11:
barrel jack

MaxPotapov, 2019-09-24:
same 5.3.1

igorpecovnik, 2019-10-17:
Closing issues that are not related to build engine. There are more than one topics already open and Armbian anyway has no resources to deal with this:

https://forum.armbian.com/topic/11340-banana-pi-m64/
https://forum.armbian.com/topic/10396-banana-pi-m64-kernel-error/

closed 2019-10-17 by igorpecovnik