armbian/build #1562: [odroidc1] I/O Accounting support

ottuzzi, 2019-09-16:
Yesterday I tried to run iotop on my Odroid C1 and the attached screenshot show the error:

[image]

It looks like we are missing some kernel configuration items.
Sorry to open an issue and not contributing the fix but in these days I have no time to work on it.

Thanks in advance
Ciao
Piero

alpe12, 2019-10-01:
https://github.com/armbian/build/issues/1465
It is working here. Maybe you are using an old version?

ottuzzi, 2019-10-03:
Hi,

  I did an apt update && apt upgrade yesterday but I'm still on kernel 5.1.15.
I saw there were updates and latest version in linux-odroidc1-next.config is 5.3.1.
I see the latest kernel is in the beta repos... I'll enable it, test it and report here.

*Ciao
Piero

ottuzzi, 2019-10-03:
Updated my odroidc1 to latest nightly `5.98.191003` (kernel `5.3.3-odroidc1`) everything seems fine.
iotop is working as expected.

Closing

*Ciao
Piero

closed 2019-10-03 by ottuzzi