armbian/build #1568: kernel rebuild breaks on patches because git stage is not clean

temoto, 2019-09-18:
Hello. Thank you for Armbian, your work is greatly appreciated.

I'm playing with kernel settings for some board and I noticed that after successful build, another `compile.sh` run will fail to apply patches, and consequently kernel make breaks in random places. I need following command line, it may work fine with different options:

`./compile.sh BOARD=orangepilite BRANCH=next KERNEL_ONLY=yes KERNEL_KEEP_CONFIG=yes BUILD_KSRC=no BSPFREEZE=yes IGNORE_UPDATES=yes`

I suspect root cause is `(cache/sources/linux-mainline/linux-4.19.y) git status` shows a lot of files (probably all files modified by patches) as staged for commit by `git add`

This magickery fixes rebuild for me:

`( cd cache/sources/linux-mainline/linux-4.19.y ; sudo git reset ; sudo git checkout . ; sudo git clean -dx -f )`

It's fast and no trouble, but to get here, I spent whole night recompiling with different kernel config trying to chase wrong problem. So maybe it helps to somebody else. Or maybe I'm using wrong rebuild compile.sh invocation?

Have a great time.

igorpecovnik, 2019-09-18:
Drop IGNORE_UPDATES=yes

temoto, 2019-10-06:
@igorpecovnik thank you. I suspect that updates break ccache, though. Could you suggest compile invocation for fast rebuilds?

igorpecovnik, 2019-10-06:
Fast rebuilds is only possible with ccache ... now if you change the code, certain parts will be rebuild. But you have to change the code properly. Each time you run the script, sources are reset and patches build on top of that. After that you can add your own patches which will change the code ... this works fine. We use this every day and there are no troubles with the ccache implementation. Its not something we would change in any way.

temoto, 2019-10-06:
Sorry, I didn't mean "updates break ccache" in fault sense, just that compilation takes too long after updates, so I suspect that updates change something which ccache properly recognize as cache miss.

Thanks for quickest responses.

igorpecovnik, 2019-10-06:
Check source file permissions, 777 will make compilation slow. Have you tried to compile manually, outside the script?

temoto, 2019-10-06:
I'm building Armbian inside your kindly provided Vagrant (because OSX insensitive filesystem), permissions are more or less consistent 0664.

Maybe I'm missing something, to compile manually I modified `lib/compilation.sh` to acquire the long make invocation at the bottom.

Scenario: optimising kernel config.

`git clean` + `compile.sh IGNORE_UPDATES=yes`: 9 minutes every time.

`regular make`: 30 seconds - basically I'm looking for how to make compile.sh only run this complex line below.

`CCACHE_BASEDIR=/home/vagrant/armbian/cache/sources/linux-mainline/linux-4.19.y PATH=/home/vagrant/armbian/cache/toolchains/gcc-linaro-7.4.1-2019.02-x86_64_arm-linux-gnueabihf/bin:/usr/lib/ccache:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/snap/bin time make -j3 ARCH=arm 'CROSS_COMPILE=ccache arm-linux-gnueabihf-' LOCALVERSION=-sunxi zImage modules dtbs`

igorpecovnik, 2019-10-07:
Hmm. Perhaps this Vagrant is causing troubles ... Don't use it. I can assure you that native works as expected.

I don't know if ccache was working properly in the past but we were doing some changes here lately @zhangn1985 @tianyuanhao any ideas? Was ccache working/works as it should under Vagrant/docker?

tianyuanhao, 2019-10-08:
confirmed

zhangn1985, 2019-10-08:
for docker:
```
root@14051c963b50:~/.ccache/tmp# ccache -s
cache directory                     /root/.ccache
primary config                      /root/.ccache/ccache.conf
secondary config      (readonly)    /etc/ccache.conf
stats zero time                     Thu Mar 14 04:13:15 2019
cache hit (direct)                735332
cache hit (preprocessed)          189077
cache miss                        248366
cache hit rate                     78.82 %
called for link                   376735
called for preprocessing           23066
multiple source files                 34
compile failed                        11
preprocessor error                     1
unsupported source language          316
unsupported compiler option          198
unsupported code directive           675
no input file                     150912
cleanups performed                   223
files in cache                    168224
cache size                           2.8 GB
max cache size                       5.0 GB
```

looks good.

temoto, 2019-10-08:
Sorry for misunderstanding. ccache is definitely working as it should. But it's less useful for rapid kernel config tuning to wait for `compile.sh` to reset, patch, recompile (even with ccache) everything when almost none files changed (so there is no need to invoke ccache

If I understand everything correctly, then `compile.sh` mandatory kernel clean+patch is the main obstacle to fast compilation. In development cycle, it should not touch network for updates and should only apply patches once. I'm looking for a clean way to do this, preferably keeping `compile.sh` great job of setting up environment.

tianyuanhao, 2019-10-08:
It works fine in Vagrant box.

```
# ./compile.sh BOARD=pcduino3 BRANCH=next KERNEL_ONLY=yes KERNEL_KEEP_CONFIG=yes BUILD_KSRC=no BSPFREEZE=yes USE_CCACHE=no
[ o.k. ] Runtime [ 23 min ]

# ./compile.sh BOARD=pcduino3 BRANCH=next KERNEL_ONLY=yes KERNEL_KEEP_CONFIG=yes BUILD_KSRC=no BSPFREEZE=yes USE_CCACHE=yes
[ o.k. ] Runtime [ 5 min ]
```

igorpecovnik, 2019-10-08:
>  In development cycle, it should not touch network for updates and should only apply patches once. 

What you are saying is that if you recompile kernel without changes its fine, runs as expected, but if you change one C file it ignores ccache altogether?

temoto, 2019-10-08:
If I change one C file or kernel config, kernel `make` takes 30 seconds, while `compile.sh` takes 9-12 minutes and the reason (I think) is recreating clean source tree and ccache-ing same just delete object files again. This approach is great and very desirable for CI and release builds but not good for development.

igorpecovnik, 2019-10-08:
Got it. I am really not sure if this worked well in the past ... need to check later. It is also possible that we break this feature by adding some other feature. Use IGNORE_UPDATES=yes and try removing this:
https://github.com/armbian/build/pull/1488/files Perhaps we just need to add this under the IGNORE_UPDATES switch?

zhangn1985, 2019-10-08:
no use.  git checkout will reset modified files and git clean will remove untracked files.  you have to add both under ignore updates. 

you have to check ignore updates when changed == true. https://github.com/armbian/build/blob/b6706ec67a1516ad356a2e76fa0050d0c80dedaa/lib/general.sh#L267

zhangn1985, 2019-10-08:
and if remote do not update.  ignore updates is conflict to force checkout. https://github.com/armbian/build/blob/b6706ec67a1516ad356a2e76fa0050d0c80dedaa/lib/general.sh#L280

you need to consolidate these two variables.

zhangn1985, 2019-10-08:
https://github.com/armbian/build/blob/9b7785a12fdd1fe7d7f42f055dde848d00bb6c52/lib/main.sh#L47

zhangn1985, 2019-10-08:
FORCE_CHECKOUT is good when need to build multiple targets.

lanefu, 2020-01-26:
@temoto did the solutions provided by zhang1985 help?   can we close?

temoto, 2020-01-26:
@lanefu Hi. Sorry for hang issue. What Zhang says is true, of course, both git clean and checkout must be skipped under IGNORE_UPDATES. There was no solution in ready-to-use sense. compile with IGNORE_UPDATES twice is broken.

Soon I will return to task requiring lots of armbian kernel rebuilds and will send you patch fixing this if I can.

lanefu, 2020-01-26:
closing. we can re-open if needed

closed 2020-01-26 by lanefu