armbian/build #1574: Upload GIT (deb)

150balbes, 2019-09-20:
How to add a package to the download is clear. Is there a way to remove packages from an existing repository ? For example. I sent kernel packages, eventually a new version appeared, how can I remove the old one (so that it does not take place in the turnips) ?

https://github.com/armbian/upload/pull/6

EvilOlaf, 2019-09-20:
You can prepare a PR that removes or replaces the package I guess.

150balbes, 2019-09-20:
If that's the case, then no problem. Everything is easy to manage. By the way, question. In reach a lot of old packages, why are they not removed ?

EvilOlaf, 2019-09-20:
Maybe someday you run into trouble with a new package and can use an old one as workaround until it is fixed. For Allwinner H6 boards this scenario happened several times during development. Ethernet stopped working on images with kernel > 4.18.0-rc7 for example and it took a while to fix it.
The same could happen to common packages as well.

igorpecovnik, 2019-09-21:
That is not implemented. You add a fixed package with a higher number and broken one is ignored and eventually purged out.

zhangn1985, 2019-10-17:
@igorpecovnik i find some targets marked as beta, does it means https://github.com/armbian/upload/tree/beta.armbian.com/ is for these targets?

igorpecovnik, 2019-10-17:
Yes! That lands in repository beta.armbian.com

closed 2021-01-27 by igorpecovnik