armbian/build #1577: Full Image builds broken

martinayotte, 2019-09-25:
It seems that Full Image builds are currently broken (full OPiPC buster 5.3.y) : 

[[[0;35m warn [[0m] Failed to download control file 
[[[0;32m .... [[0m] downloading using http(s) network [[[0;33m buster-cli-armhf.3194e810e2b8117e7c1ec2fc28e4f313.tar.lz4 [[0m]
[[[0;32m o.k. [[0m] ... remote not found [[[0;33m Creating new rootfs cache for buster [[0m]
[[[0;32m o.k. [[0m] Installing base system [[[0;33m Stage 1/2 [[0m]
I: Retrieving InRelease 
I: Checking Release signature
E: Release signed by unknown key (key id DCC9EFBF77E11517)
[[[0;31m error [[0m] ERROR in function create_rootfs_cache [[[0;33m debootstrap.sh:147 [[0m]
[[[0;31m error [[0m] Debootstrap base system first stage failed 
[[[0;32m o.k. [[0m] Process terminated 

Any ideas ?

martinayotte, 2019-09-25:
Oupps ! New build, new errors : 

[ o.k. ] Building package [ linux-buster-root-dev-orangepipc ]
[ o.k. ] Building desktop package [ armbian-buster-desktop_5.97_all ]
[ o.k. ] Starting rootfs and image building process for [ orangepipc buster ]
[ o.k. ] Checking for local cache [ buster-cli-armhf.319...313.tar.lz4 ]
[ .... ] searching on servers 
Exception: [AbstractOptionHandler.cc:69] errorCode=28 We encountered a problem while processing the option '--download-result'.
  -> [OptionHandlerImpl.cc:444] errorCode=1 download-result must be one of the following:'default' 'full' 
Usage:
 --download-result=OPT        This option changes the way "Download Results"
                              is formatted. If OPT is 'default', print GID,
                              status, average download speed and path/URI. If
                              multiple files are involved, path/URI of first
                              requested file is printed and remaining ones are
                              omitted.
                              If OPT is 'full', print GID, status, average
                              download speed, percentage of progress and
                              path/URI. The percentage of progress and
                              path/URI are printed for each requested file in
                              each row.

                              Possible Values: default, full
                              Default: default
                              Tags: #advanced
[ warn ] Failed to download control file 
[ .... ] downloading using http(s) network [ buster-cli-armhf.3194e810e2b8117e7c1ec2fc28e4f313.tar.lz4 ]
Exception: [AbstractOptionHandler.cc:69] errorCode=28 We encountered a problem while processing the option '--download-result'.
  -> [OptionHandlerImpl.cc:444] errorCode=1 download-result must be one of the following:'default' 'full' 
Usage:
 --download-result=OPT        This option changes the way "Download Results"
                              is formatted. If OPT is 'default', print GID,
                              status, average download speed and path/URI. If
                              multiple files are involved, path/URI of first
                              requested file is printed and remaining ones are
                              omitted.
                              If OPT is 'full', print GID, status, average
                              download speed, percentage of progress and
                              path/URI. The percentage of progress and
                              path/URI are printed for each requested file in
                              each row.

                              Possible Values: default, full
                              Default: default
                              Tags: #advanced
[ o.k. ] ... remote not found [ Creating new rootfs cache for buster ]
[ o.k. ] Installing base system [ Stage 1/2 ]
I: Retrieving InRelease 
I: Checking Release signature
E: Release signed by unknown key (key id DCC9EFBF77E11517)
[ error ] ERROR in function create_rootfs_cache [ debootstrap.sh:147 ]
[ error ] Debootstrap base system first stage failed 
[ o.k. ] Process terminated 
[ o.k. ] Unmounting [ /disk2/home/martin/git-work/igorpecovnik-armbian/build/.tmp/rootfs-dev-orangepipc-buster-no-no/ ]
[ error ] ERROR in function unmount_on_exit [ image-helpers.sh:66 ]
[ error ] debootstrap-ng was interrupted 
[ o.k. ] Process terminated

igorpecovnik, 2019-09-26:
One download server is down due to maintenance / possible disk failure. Perhaps related ... investigation.

igorpecovnik, 2019-09-29:
I was not able to recreate by going from a clean build.

> aria2c: command not found

This tells some troubles with host. Perhaps dependency installation was not successful.

closed 2019-09-29 by igorpecovnik