armbian/build #1578: /bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)

TelDragon, 2019-09-27:
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)

At build time, it is not known that the DEST_LANG parameter in "config-default.conf" is invalid.

From which version is this parameter discarded? How can I enable it later?

igorpecovnik, 2019-09-27:
This is unlikely to be a problem of a build script. More as a problem of your host.

TelDragon, 2019-09-27:
This is the case. When I built it earlier, the config-default.conf file was selected with the parameter "DEST_LANG". I have not paid attention to it in the later period.

I didn't observe it until after the warning appeared. Found that the latest build script is out of this parameter, the document does not have a description of this parameter.

igorpecovnik, 2019-09-27:
> latest build script is out of this parameter

Its present by default:
https://github.com/armbian/build/blob/master/config/templates/config-example.conf#L12

If your system is broken or altered you might create the problem ... I tried on a clean build and can't see any troubles / warnings of this kind. Please provide a log where this happens starting from a clean build.

Averell7, 2019-10-01:
I intended to set an issue about this question. In Debian Buster with Armbian Linux 4.4.182-rockchip64 it is impossible to change the locale which is locked to "C" in /etc/environment.
It will never be possible to change the locale of the console, either physical or SSL, as long as this line is there. armbian-config and dpkg-reconfigure has  no effect. All locale parameters are sticked to C.
The result with a French keyboard is the most strange I have ever seen. 
When you type : ùc, you get : ccccccccc
éèçà just don't appear at all. Same thing with cat 
Please, please, please, remove :
LC_ALL=C 
from /etc/environment. 
Perhaps it has an impact on the problem you submitted, TelDragon ?

TelDragon, 2019-10-12:
This problem still occurs with the new environment. Looking for other reasons
```
I: Extracting bsdutils...
I: Extracting libblkid1...
I: Extracting libfdisk1...
I: Extracting libmount1...
I: Extracting libsmartcols1...
I: Extracting libuuid1...
I: Extracting mount...
I: Extracting util-linux...
I: Extracting liblzma5...
I: Extracting zlib1g...
[ o.k. ] Installing base system [ Stage 2/2 ]
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
I: Installing core packages...
I: Unpacking required packages...
I: Unpacking libacl1:armhf...
I: Unpacking libattr1:armhf...
```

TelDragon, 2019-10-12:
Verified, just introduce userpatches/lib.config
Still using the official docs example

I have deprecated this way to install additional software, using userpatches/customize-image.sh and then installing it inside the chroot.

TelDragon, 2019-10-12:
Also found a small problem

```
[ .... ] Installing [ linux-dtb-next-sunxi_5.98_armhf.deb ]
E: Unable to locate package wireguard-tools
[ .... ] Installing [ linux-stretch-root-next-t1000p_5.98_armhf.deb ]
```
Here code again

/build/lib/distributions.sh
```
 # install wireguard tools
        chroot "${SDCARD}" /bin/bash -c "apt -y -qq install wireguard-tools"
```

igorpecovnik, 2019-10-12:
> Isn't this an important piece of software?

no if you don't use it. its for controlling vpn

closed 2019-10-13 by TelDragon