armbian/build #1582: Logrotate not working properly in Armbian 4.4.184

Averell7, 2019-10-01:
The ramlog feature in Armbian, is a very good idea, but it is not yet properly implemented. 

1) There is a bug in /usr/lib/armbian/armbian-ramlog
In the syncToDisk function, the parameter "--delete" is a nonsense in the rsync command, line 50. It makes /var/log.hdd an exact copy of /var/log, deleting all rotated files, and making log.hdd totally useless. It is an obvious mistake. The "--delete" parameter may be useful in the syncFromDisk function, in case of restart, and at least does no harm there, but it MUST be removed from the syncToDisk function.  

2) Once this is corrected, log.hdd is better, but there is still an issue : the rotated files overlap, because the effect of the rotation is not mirrored to /var/log. I have a log rotated daily, but each rotated file contains 3 or 4 days. 
This can be solved easily by the following process : 

- ramlog write  (to update log.hdd before rotation)
- logrotate
- ramlog postrotate (which will copy to /var/log the modified files of /ram/log/hdd.

This works perfectly here and is implemented by the two following modifications : 
in /usr/lib/armbian/armbian-ramlog, adding a "postrotate" action in the case statement, in the end : 
```
postrotate)
    cd /var/log.hdd/
    find . -type f -print | grep -E -v "(\.gz|\.[0-9])" | while IFS= read -r  file
    do
        dest="/var/log/$file"
        cat $file > $dest
        echo "copy /var/log.hdd/$file  =======> $dest"
    done
        ;;
```
(the echo is for debug only)
Modify the logrotate.service file : 

```
[Service]
Type=oneshot
ExecStartPre=/usr/lib/armbian/armbian-ramlog write
ExecStart=/usr/sbin/logrotate /etc/logrotate.conf
ExecStartPost=/usr/lib/armbian/armbian-ramlog postrotate

```
On a working system, it is better to copy this file from /lib/systemd/system   to /etc/systemd/system to prevent possible overwriting  during an update.

lanefu, 2019-10-07:
Hi @Averell7 thanks for your analysis.  Would you be able to provide a PR?

files above should reside here

https://github.com/armbian/build/tree/master/packages/bsp/common

Averell7, 2019-10-07:
I have never done that, but I will try. 
I must add an explanation : I wrote above : _I use find because rsync is not well featured for exclusion_, that's not true, but the reason was not that, I remember now. The problem with rsync is that it creates new files, and several programs don't like it and stop logging, even if the file name is the same. And when the file is renamed, they continue logging in the renamed file. That's why cat source > destination was better because destination is overwritten, but (I guess) keeps the same id.

Averell7, 2019-10-07:
I think I see what I have to do. The system gives a warning : 
_You’re creating a file in a project you don’t have write access to. Submitting a change will create the file in a new branch in your fork Averell7/build, so you can send a pull request._ 
Is that OK for you ?

igorpecovnik, 2019-10-07:
Yes. Now edit the file and commit to your branch. Then click "Create pull request"
https://docs.armbian.com/Process_Contribute/

botfap, 2019-10-07:
Excellent, Ive been ignoring this for a while!

Just tested your PR #1588 and have some probably ignorant questions:

1. Would it not be simpler to trigger the postrotate function with a line at the end of /etc/cron.daily/logrotate instead of a new unit file? e.g.
```
#!/bin/sh

# Clean non existent log file entries from status file
cd /var/lib/logrotate
test -e status || touch status
head -1 status > status.clean
sed 's/"//g' status | while read logfile date
do
    [ -e "$logfile" ] && echo "\"$logfile\" $date"
done >> status.clean
mv status.clean status

test -x /usr/sbin/logrotate || exit 0
/usr/lib/armbian/armbian-ramlog write
/usr/sbin/logrotate /etc/logrotate.conf
/usr/lib/armbian/armbian-ramlog postrotate
```

2. Is syncToDisk (rsync --delete) not still being called for stop and write? If so, is it not the case that a user stopping or restarting the armbian-ramlog service is also going to wipe out the rotated logs anyway?

3. Is it worth adding mysql to the RecreateLogs function? I often get mysql start failures due to missing log directory after a crash

4. In syncFromDisk should we also not exclude *.xz files in addition to *.gz (cant remember where from but I have had a few .xz files synced back into the ramlog partition

botfap, 2019-10-08:
Fix tested & merged

closed 2019-10-08 by botfap

lanefu, 2019-10-10:
[AR-6]

igorpecovnik, 2019-10-12:
[AR-9] tew

Miro-sk, 2021-02-06:
Hello,
I am using Armbian 21.02.1 Focal with Linux 5.10.12-sunxi. 
I use the logrotate functionality but I have a problem that the /var/log/syslog file is still growing (even if logrotate is run daily). As Averell7 writes above, the logrotate process takes place in 3 steps (see /etc/systemd/system/logrotate.service -> [Service] section):

1) ramlog write (to update log.hdd before rotation) -> ExecStartPre =/usr/lib/armbian/armbian-ramlog write
2) logrotate -> ExecStart =/usr/sbin/logrotate /etc/logrotate.conf
3) ramlog postrotate (which will copy to/var/log the modified files of /var/log.hdd) -> ExecStartPost =/usr/lib/armbian/armbian-ramlog postrotate

I think step 3 is wrong. 
I assume the following steps during logrotate:
1) ramlog write (to update log.hdd before rotation) - it is ok
2) the system writes another data to /var/log/syslog (at this moment /var/log/syslog <> /var/log.hdd/syslog)- that's ok 
3) logrotate is performed (/var/log/syslog is renamed to /var/log/syslog.1, an empty /var/log/syslog is created, the system starts writing to the new empty /var/log/syslog). - it is OK
4) the system writes data to new /var/log/syslog (at this step /var/log/syslog is not empty) - that's ok
5) the ramlog postrotate is started.
The command "cat $ file> $ dest" overwrites data in new /var/log/syslog with data from the old /var/log.hdd/syslog - so data written to syslog in step 4 is lost - this is a bug in my opinion

Another consequence of ramlog postrotate is that the new file /var/log/syslog is filled with the data that was written to syslog in the past (so the /var/log/syslog file is still growing).
Why is necessary to copy the old data from /var/log.hdd/syslog to to the new file/var/log/syslog?
 
Thanks for the explanation.

Averell7, 2021-02-06:
In your explanation, point 3 is wrong. Rotating /var/log/syslog is useless, since it is in RAM. Just shut down your device and all rotated files are lost. Logrotate must rotate /var/log.hdd/syslog , creating a new empty file which overwrites in step 5 the /var/log/syslog file, which will be empty in the end of the process. 
You are right to note that some log lines may be lost in the process (your step 4). If you find a way to improve that, explain. But is your syslog file is growing and contains more than a day of data, the responsible is your logrotate configuration which does not rotate the right file (log, instead of log.hdd).

Miro-sk, 2021-02-06:
Thank you for your reply. I use Armbian Focal from https://www.armbian.com/orange-pi-zero/, where the "default logrotate configuration" rotates files in /var/log (not /var/log.hdd). So I think a lot of users have the same problem as me.

My opinion is to change
/etc/systemd/system/logrotate.service -> [Service] section as follows:

[Service]
Type=oneshot
# ExecStartPre=/usr/lib/armbian/armbian-ramlog write
ExecStart=/usr/sbin/logrotate /etc/logrotate.conf
# ExecStartPost=/usr/lib/armbian/armbian-ramlog postrotate
ExecStartPost=/usr/lib/armbian/armbian-ramlog write

There are no lost entries in the syslog in this solution.

I think the possibility of data loss is the same as in your solution.

What do you think?

Averell7, 2021-02-06:
Just test and see. But I think it was the previous situation which didn't work. I don't have the code here but I guess the syslog file (and others) will continue growing. Or you would have to double the logrotate configuration and rotate log.hdd and log. 
For the test to be significant, you must let your device running without stopping if for several days. 

About Armbian Focal not being updated, I just don't know the situation since I am not a developper of the project.