armbian/build #1592: MAINLINE_MIRROR="tuna"   Invalid argument

TelDragon, 2019-10-11:
ps 

MAINLINE_MIRROR select mainline mirror of linux-stable.git
set to google to use mirror provided by Google, the same as USE_MAINLINE_GOOGLE_MIRROR=yes.
set to tuna to use mirror provided by tsinghua university.
leave it empty to use offical git.kernel.org, it may be very slow for mainland china users.

Clone a new library
Preferences  tuna
The process is still  
`
/usr/lib/git-core/git-remote-https origin https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git
/usr/lib/git-core/git fetch-pack --stateless-rpc --stdin --lock-pack --thin --depth=1 https://git.kernel.org/pub/scm/linux/
/usr/lib/git-core/git --shallow-file /home/ubuntu/build/cache/sources/linux-firmware-git/.git/shallow.lock index-pack --std
`

TelDragon, 2019-10-11:
I suddenly found out  

~/build/packages/
extras  Folder does not exist    same   firmware.sh  file does not exist

TelDragon, 2019-10-11:
It seems that the reason has been found.
According to the ~/build/lib/configuration.sh script
Find

`
# used by multiple sources - reduce code duplication
[[ $USE_MAINLINE_GOOGLE_MIRROR == yes ]] && MAINLINE_MIRROR=google
case $MAINLINE_MIRROR in
        google) MAINLINE_KERNEL_SOURCE='https://kernel.googlesource.com/pub/scm/linux/kernel/git/stable/linux-stable' ;;
        tuna) MAINLINE_KERNEL_SOURCE='https://mirrors.tuna.tsinghua.edu.cn/git/linux-stable.git' ;;
        *) MAINLINE_KERNEL_SOURCE='git://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git' ;;
esac
`

curl  https://mirrors.tuna.tsinghua.edu.cn/git/linux-stable.git            404 ~~

Will not be abandoned by Tsinghua University.

igorpecovnik, 2019-10-11:
Check if you can find the reason for this.

TelDragon, 2019-10-11:
Local Testing The library is cloned normally.

igorpecovnik, 2019-10-11:
No, we want to have China mainland mirror. Just don't use MAINLINE_MIRROR="tuna" until this is resolved.

@mzhboy

TelDragon, 2019-10-11:
Maybe it's my mistake or other reasons. Is it before linux-firmware.git China has never had a mirror. Only Google and the official git.kernel.org

Because I browsed the tuna and did not find support for linux-firmware.git.

closed 2019-10-11 by TelDragon