armbian/build #1596: Marvell Armada thermal driver

taggart, 2019-10-18:
I am looking for support for temperature sensors in armbian on espressobin. I looked in

https://github.com/armbian/build/tree/master/patch/kernel/mvebu64-*

but didn't see anything. I see that the vendor has something,

https://github.com/MarvellEmbeddedProcessors/linux-marvell/blob/master/drivers/thermal/armada_thermal.c

and that something appears to have gone upstream,

https://github.com/torvalds/linux/blob/master/drivers/thermal/armada_thermal.c

but I don't know if these include support for the hardware in the espressobin.
Do you know when this will work in armbian?

Thanks for your work!

igorpecovnik, 2019-10-19:
> but didn't see anything. I see that the vendor has something,
> I see that the vendor has something

This means feature is not ported, implemented or broken. I don't know without research.

> Do you know when this will work in armbian?

When its fixed. Perhaps this month, year or never. This is the case in amateur based development - if someone finds time. R&D budget is exactly zero.

BTW. This is [Armbian Linux build script](https://github.com/armbian/build) issue tracker. Kernel issues regarding Espressobin are discussed here: https://forum.armbian.com/topic/4089-espressobin-support-development-efforts The topic is huge and perhaps your answers are already there.

taggart, 2019-10-22:
Sorry is this came off sounding demanding, I wasn't asking for the work to be done, I was just wondering when upstream changes might land in the tree.
Thanks for the pointer to the forum, I'll look there

closed 2019-10-22 by taggart