armbian/build #1599: ERROR in function download_and_verify on running compile.sh

dbaba, 2019-10-21:
I'm trying to build an Armbian image on Vagrant with ubuntu/bionic64 box but always got stuck on an error shown below while running compile.sh.

Is the GCC compilers archive file corrupted or is there something wrong in the compiler.sh script?

```
vagrant@ubuntu-bionic:/vagrant$ sudo ./compile.sh 
[ o.k. ] Using config file [ /vagrant/userpatches/config-example.conf ]
[ o.k. ] Preparing [ host ]
[ o.k. ] Build host OS release [ bionic ]
[ o.k. ] Syncing clock [ host ]
[ o.k. ] Checking for external GCC compilers 
[#196266 0B/0B CN:1 DL:0B][ .... ] downloading using torrent network [ gcc-linaro-aarch64-none-elf-4.8-2013.11_linux.tar.xz ]             
[ error ] ERROR in function download_and_verify [ general.sh:995 ]                                                                        
[ error ] verification failed 
[ o.k. ] Process terminated 
```

igorpecovnik, 2019-10-21:
Try running with additional parameter USE_TORRENT="no"

dbaba, 2019-10-21:
Thanks for the reply. But still not working. The failure seemed to be a kind of PGP error but not sure what it was.

```
vagrant@ubuntu-bionic:/vagrant$ rm -fr ./cache/
vagrant@ubuntu-bionic:/vagrant$ sudo ./compile.sh USE_TORRENT="no"
[ o.k. ] Using config file [ /vagrant/userpatches/config-example.conf ]
[ o.k. ] Command line: setting USE_TORRENT to [ no ]
[ o.k. ] Preparing [ host ]
[ o.k. ] Build host OS release [ bionic ]
[ o.k. ] Syncing clock [ host ]
[ o.k. ] Checking for external GCC compilers 
[#22aaa1 0B/0B CN:1 DL:0B][ .... ] downloading using http(s) network [ gcc-linaro-aarch64-none-elf-4.8-2013.11_linux.tar.xz ]             
[#e2ed86 23MiB/24MiB(96%) CN:1 DL:1.9MiB]                                                                                                 
[ error ] ERROR in function download_and_verify [ general.sh:995 ]
[ error ] verification failed 
[ o.k. ] Process terminated 
```

I looked into the general.sh script and I tried the following command related to the binary verification and got a PGP error.
How can I install the public key for the PGP verification? Or it's out of the issue scope?
Anyway, any other suggestions are appreciated.

```
vagrant@ubuntu-bionic:/vagrant$ gpg --homedir ./cache/.gpg/ --no-permission-warning --verify --trust-model always -q ./cache/toolchains/gcc-linaro-aarch64-none-elf-4.8-2013.11_linux.tar.xz.asc 
gpg: assuming signed data in './cache/toolchains/gcc-linaro-aarch64-none-elf-4.8-2013.11_linux.tar.xz'
gpg: Signature made Tue Nov 19 09:00:03 2013 UTC
gpg:                using RSA key D47877808F427EAF
gpg: Can't check signature: No public key
```

igorpecovnik, 2019-10-21:
> Anyway, any other suggestions are appreciated.
> How can I install the public key for the PGP verification? Or it's out of the issue scope?

Everything shell be installed automatically ... but Vagrant build way is not well tested. I don't use it. Try native or Docker way of building. That must work.

Perhaps @tianyuanhao have an idea?

dbaba, 2019-10-22:
Thanks.
I found this error was caused by my own Vagrantfile, which didn't have a valid provisioning script.
I copied the project's Vagrantfile from the older tag (https://github.com/armbian/build/blob/v2017.11/Vagrantfile), updated box to bionic in the file, `vagrant up`, ran `sudo ./compile.sh` on `/home/ubuntu/armbian`after `vagrant ssh`, and the GPG error was gone in the end.

I read the [instruction document](https://docs.armbian.com/Developer-Guide_Using-Vagrant/) in order to build Armbian on vagrant host but the latest master branch doesn't have a required file (Vagrantfile).
Is this way (vagrant host steps) no longer supported?

tianyuanhao, 2019-10-22:
It was moved to `config/templates/Vagrantfile`.

dbaba, 2019-10-22:
@tianyuanhao Thanks!

closed 2019-10-22 by dbaba