armbian/build #1604: /boot/dtb symbolic link missing in recent builds

martinayotte, 2019-10-25:
Hi Igor,
I faced the issue of /boot/dtb symbolic link missing in recent builds, I had to create it manually ...
In our script, for example for sunxi-dev with preview of 5.4.y, I've found general-packaging-4.20-dev.patch, but it doesn't seem to do the job anymore ... (I seen that also in recent rk3399 builds with 5.4.y)
Do you have any clues ?

igorpecovnik, 2019-10-26:
Can only confirm that I noticed the same yesterday for the first time.

martinayotte, 2019-11-09:
I've taken some look at the issue, it seems that those errors are the source of the problem : 

```
./scripts/package/builddeb: 45: ./scripts/package/builddeb: [[: not found
./scripts/package/builddeb: 59: ./scripts/package/builddeb: [[: not found
dpkg-deb: building package 'linux-headers-dev-sunxi64' in '../linux-headers-dev-sunxi64_5.98_arm64.deb'.
./scripts/package/builddeb: 45: ./scripts/package/builddeb: [[: not found
./scripts/package/builddeb: 59: ./scripts/package/builddeb: [[: not found
dpkg-deb: building package 'linux-dtb-dev-sunxi64' in '../linux-dtb-dev-sunxi64_5.98_arm64.deb'.
./scripts/package/builddeb: 45: ./scripts/package/builddeb: [[: not found
./scripts/package/builddeb: 59: ./scripts/package/builddeb: [[: not found
dpkg-deb: building package 'linux-image-dev-sunxi64' in '../linux-image-dev-sunxi64_5.98_arm64.deb'.
```

martinayotte, 2019-11-09:
I think I've started to figure out : In Torvald's tree, the CONFIG_SHELL definition in root Makefile as change from : 
```
CONFIG_SHELL := $(shell if [ -x "$$BASH" ]; then echo $$BASH; \
	  else if [ -x /bin/bash ]; then echo /bin/bash; \
	  else echo sh; fi ; fi)
```
to : 
```
CONFIG_SHELL := sh
```
Which is produced the "[[" errors ...
I will try to fix that later by changing "[[" to "[", will see if it work.

zhangn1985, 2019-11-27:
@igorpecovnik please take a look, because many devices are updated to 5.4.y

missing postinst file in both dtb.deb and header.deb

igorpecovnik, 2019-11-27:
On it.

martinayotte, 2019-11-27:
During my preview of 5.4.y, I've fix the issue in general-packaging-4.20.patch, but this patch has been deleted, so I've recreate new one here :
https://github.com/armbian/build/blob/master/patch/kernel/sunxi-dev/general-fix-builddeb-packaging.patch
But maybe there is something else needed that was part of previous patch ...

igorpecovnik, 2019-11-27:
I fix it differently :) Will implement it later during a day.

closed 2019-11-27 by igorpecovnik