armbian/build #1618: Impossible to remove directory with cloned repository after dockerized image build

tysonite, 2019-11-09:
After succesfull build of image, it is impossible to complete remove directory where build repository was cloned.

Sequency of commands that lead to error:
```
git clone <armbian_build_url> armbian_build
cd armbian_build
./compile.sh docker <name>
cd .. && rm -rf armbian_build
rm: cannot remove '/home/user/armbian_build/.tmp/xenial/orangepione': Permission denied
```

martinayotte, 2019-11-09:
Did you tried with "sudo" ?

tysonite, 2019-11-09:
Sure. It worked. However I expect files to be removed without switching priviledges as I run `./compile.sh` as unpriviledges user.

martinayotte, 2019-11-09:
If you look at ./compile.sh around line 37, it looks at first argument to determine if it in docker mode, if not, so normal shell, it gives an alert saying that ./compile.sh must be run as sudo.
Since I never use docker with Armbian, I don't know the exact mechanism, but it is somehow doing a sudo automatically. In those same lines, you can see also argument for "dockerpurge", which is probably the same as "sudo rm -rf " ...

tysonite, 2019-11-09:
Not sure that it `sudo`s to root. There was no request to input a password. E.g. [this line](https://github.com/armbian/build/blob/336cf390d094766168c7593058f18cc8fc8f5958/compile.sh#L111) suggests to add user to a docker group to avoid root priviledges. This has been done by me.

igorpecovnik, 2019-11-12:
AFAIK images can only be build under privileged mode. I am not sure we have other option. Yet.

closed 2019-11-12 by igorpecovnik