armbian/build #1621: armbian-truncate-logs force Logrotation before to clear logs does not work

GAS85, 2019-11-11:
* Ubuntu 4.19.62
* Cubietruck with ARM
* Armbian 5.90

Hello,
I find out that **armbian-truncate-logs** somehow does not work as expected. It should move logs to hdd and force log rotation, but somehow I have no *.gz or *.log.1 files in folder after armbian-truncate-logs starts. Nether in `/var/log` or `/var/log.hdd` Only fresh and small *.log files. Basically I loose all logs since threshold 75% of log2ram will be reached.

If I run command manually I will see *.gz files in log.hdd folder:

    sudo logrotate --force /etc/logrotate.conf

<details>
<summary>Content of folders as per now</summary>

```
# ls -lah /var/log /var/log.hdd
/var/log:
total 960K
drwxrwxr-x  15 root     syslog          4.0K Sep  6 11:53 .
drwxr-xr-x  16 root     root            4.0K Sep  6 11:54 ..
-rw-r--r--   1 root     root               0 Nov 10 04:31 alternatives.log
drwxr-xr-x   2 root     root            4.0K Aug 27 13:27 apache2
drwxr-xr-x   2 root     root            4.0K Nov  7 14:47 apt
-rw-r--r--   1 root     root            2.6K Sep  3 15:19 aptitude
-rw-rw-r--   1 vpn      vpn                0 Nov 10 04:31 aria2.log
-rw-r--r--   1 root     root               0 Nov 10 04:31 armbian-hardware-monitor.log
-rw-r-----   1 syslog   adm             291K Nov 11 10:50 auth.log
-rw-r--r--   1 root     root               0 Nov 10 04:31 backup_72de3.log
-rw-r--r--   1 root     root             19K Nov 11 10:40 battery_checker.log
-rw-r--r--   1 root     root               0 Nov 10 04:31 bootstrap.log
-rw-rw----   1 root     utmp               0 Nov 10 04:31 btmp
drwxr-xr-x   2 www-data www-data        4.0K Aug 19 20:38 cacti
drwxr-xr-x   2 root     root            4.0K Aug 19 20:30 dbconfig-common
-rw-r--r--   1 root     root               0 Nov 10 04:31 dpkg.log
-rw-r-----   1 root     adm             192K Nov 11 10:46 fail2ban.log
-rw-r--r--   1 root     root               0 Nov 10 04:31 faillog
-rw-r--r--   1 root     root               0 Nov 10 04:31 fontconfig.log
drwxr-sr-x+  2 root     systemd-journal 4.0K Jun 20 00:20 journal
-rw-r-----   1 syslog   adm                0 Nov 10 04:31 kern.log
-rw-rw-r--   1 root     utmp            286K Nov 11 10:18 lastlog
drwx------   2 root     root            4.0K Nov 11 05:21 letsencrypt
drwx------   2 root     root             16K Sep  4 10:31 lost+found
-rw-r-----   1 syslog   adm                0 Nov 10 04:31 mail.err
-rw-r-----   1 syslog   adm              341 Nov 11 00:03 mail.log
-rw-r--r--   1 minidlna minidlna        5.3K Nov 11 06:26 minidlna.log
-rw-r--r--   1 root     root               0 Nov 10 04:31 nand-sata-install.log
-rw-r--r--   1 www-data www-data        4.3K Nov 11 10:00 next-cron.log
lrwxrwxrwx   1 root     root              33 Aug 21 14:41 nextcloud.log -> /BlaBlaBla/nextcloud.log
drwxr-xr-x   2 ntp      ntp             4.0K Jul  6  2018 ntpstats
drwxr-xr-x   2 root     root            4.0K May 14 10:25 openvpn
-rw-------   1 root     root            7.8K Nov 11 10:35 php7.3-fpm.log
drwxr-x---   2 redis    redis           4.0K Nov 10 04:31 redis
drwxr-x---   3 root     adm             4.0K Sep 22 00:41 samba
-rw-r-----   1 syslog   adm             262K Nov 11 10:50 syslog
drwxr-xr-x   2 root     root            4.0K Dec 23  2017 sysstat
-rw-------   1 root     root             63K Sep 13 09:57 tallylog
drwxr-x---   2 root     adm             4.0K Aug 19 19:17 unattended-upgrades
-rw-r--r--   1 root     root               0 Nov 10 04:31 upgrade.log
-rw-rw-r--   1 root     utmp            2.7K Nov 11 10:47 wtmp

/var/log.hdd:
total 1.2M
drwxrwxr-x  15 root     syslog          4.0K Nov 11 06:25 .
drwxr-xr-x  16 root     root            4.0K Sep  6 11:54 ..
-rw-r--r--   1 root     root               0 Nov 10 04:31 alternatives.log
drwxr-xr-x   2 root     root            4.0K Nov 11 06:25 apache2
drwxr-xr-x   2 root     root            4.0K Nov  7 14:47 apt
-rw-r--r--   1 root     root            2.6K Sep  3 15:19 aptitude
-rw-rw-r--   1 vpn      vpn                0 Nov 10 04:31 aria2.log
-rw-r--r--   1 root     root               0 Nov 10 04:31 armbian-hardware-monitor.log
-rw-r--r--   1 root     root             86K Nov 11 06:25 armbian-ramlog.log
-rw-r-----   1 syslog   adm             248K Nov 11 06:25 auth.log
-rw-r--r--   1 root     root               0 Nov 10 04:31 backup_72de3.log
-rw-r--r--   1 root     root             17K Nov 11 06:20 battery_checker.log
-rw-r--r--   1 root     root               0 Nov 10 04:31 bootstrap.log
-rw-rw----   1 root     utmp               0 Nov 10 04:31 btmp
drwxr-xr-x   2 www-data www-data        4.0K Aug 19 20:38 cacti
drwxr-xr-x   2 root     root            4.0K Aug 19 20:30 dbconfig-common
-rw-r--r--   1 root     root               0 Nov 10 04:31 dpkg.log
-rw-r-----   1 root     adm             189K Nov 11 06:21 fail2ban.log
-rw-r--r--   1 root     root               0 Nov 10 04:31 faillog
-rw-r--r--   1 root     root               0 Nov 10 04:31 fontconfig.log
drwxr-sr-x+  2 root     systemd-journal 4.0K Jun 20 00:20 journal
-rw-r-----   1 syslog   adm                0 Nov 10 04:31 kern.log
-rw-rw-r--   1 root     utmp            286K Nov 10 14:12 lastlog
drwx------   2 root     root            4.0K Nov 11 05:21 letsencrypt
drwx------   2 root     root            4.0K Aug 19 19:17 lost+found
-rw-r-----   1 syslog   adm                0 Nov 10 04:31 mail.err
-rw-r-----   1 syslog   adm              341 Nov 11 00:03 mail.log
-rw-r--r--   1 minidlna minidlna        3.9K Nov 10 16:46 minidlna.log
-rw-r--r--   1 root     root               0 Nov 10 04:31 nand-sata-install.log
-rw-r--r--   1 www-data www-data        4.2K Nov 11 06:00 next-cron.log
lrwxrwxrwx   1 root     root              33 Aug 21 14:41 nextcloud.log -> /BlaBlaBla/nextcloud.log
drwxr-xr-x   2 ntp      ntp             4.0K Jul  6  2018 ntpstats
drwxr-xr-x   2 root     root            4.0K May 14 10:25 openvpn
-rw-------   1 root     root               0 Nov 10 04:31 php7.3-fpm.log
drwxr-x---   2 redis    redis           4.0K Nov 10 04:31 redis
drwxr-x---   3 root     adm             4.0K Sep 22 00:41 samba
-rw-r-----   1 syslog   adm                0 Nov 11 06:25 syslog
-rw-r-----   1 syslog   adm             229K Nov 11 06:25 syslog.1
drwxr-xr-x   2 root     root            4.0K Dec 23  2017 sysstat
-rw-------   1 root     root             63K Sep 13 09:57 tallylog
drwxr-x---   2 root     adm             4.0K Aug 19 19:17 unattended-upgrades
-rw-r--r--   1 root     root               0 Nov 10 04:31 upgrade.log
-rw-rw-r--   1 root     utmp            1.5K Nov 10 14:13 wtmp
```

</summary>

igorpecovnik, 2019-11-12:
What is your /etc/logrotate.conf

GAS85, 2019-11-12:
Never touch it since installation.

```
# see "man logrotate" for details
# rotate log files weekly
weekly

# use the syslog group by default, since this is the owning group
# of /var/log.hdd/syslog.
su root syslog

# keep 4 weeks worth of backlogs
rotate 4

# create new (empty) log files after rotating old ones
create

# uncomment this if you want your log files compressed
compress

# packages drop log rotation information into this directory
include /etc/logrotate.d

# no packages own wtmp, or btmp -- we'll rotate them here
/var/log.hdd/wtmp {
    missingok
    monthly
    create 0664 root utmp
    rotate 1
}

/var/log.hdd/btmp {
    missingok
    monthly
    create 0660 root utmp
    rotate 1
}

# system-specific logs may be configured here
```

igorpecovnik, 2020-01-02:
Can't reproduce on latest image. It seems we have fixed this problem.

closed 2020-01-02 by igorpecovnik