armbian/build #1626: Build fails on master branch, works on tag sunxi-5.3

Stadicus, 2019-11-20:
Building from master branch (commit https://github.com/armbian/build/commit/d184b0393038be2bf29d78fd9cf469e7a26b5f0c) fails to build with the following configuration.

Building from tag https://github.com/armbian/build/releases/tag/sunxi-5.3 works fine.

```sh
./compile.sh docker BOARD=rockpro64 KERNEL_ONLY=no KERNEL_CONFIGURE=no BUILD_MINIMAL=yes BUILD_DESKTOP=no RELEASE=bionic BRANCH=default WIREGUARD=no PROGRESS_LOG_TO_FILE=yes

[ o.k. ] Using config file [ /mnt/hdd/github/bitbox-base/armbian/armbian-build/userpatches/config-docker.conf ]
[ .... ] Using existed a armbian Docker container 
[ o.k. ] Running the container 
[ o.k. ] Using config file [ /root/armbian/userpatches/config-example.conf ]
[ o.k. ] Command line: setting BOARD to [ rockpro64 ]
[ o.k. ] Command line: setting KERNEL_ONLY to [ no ]
[ o.k. ] Command line: setting KERNEL_CONFIGURE to [ no ]
[ o.k. ] Command line: setting BUILD_MINIMAL to [ yes ]
[ o.k. ] Command line: setting BUILD_DESKTOP to [ no ]
[ o.k. ] Command line: setting RELEASE to [ bionic ]
[ o.k. ] Command line: setting BRANCH to [ default ]
[ o.k. ] Command line: setting WIREGUARD to [ no ]
[ o.k. ] Command line: setting PROGRESS_LOG_TO_FILE to [ yes ]
[ o.k. ] Preparing [ host ]
[ o.k. ] Build host OS release [ bionic ]
[ o.k. ] Running in container [ docker ]
[ warn ] apt-cacher is disabled in containers, set NO_APT_CACHER=no to override 
[ o.k. ] Checking for external GCC compilers 
[ .... ] downloading using torrent network [ gcc-arm-8.3-2019.03-x86_64-arm-linux-gnueabihf.tar.xz ]
[ o.k. ] Verified [ MD5 ]
[ .... ] decompressing 
[ .... ] gcc-arm-8.3-2019.03-x86_64-arm-linux-gnueabihf.tar.xz: 54.2MiB [12.8MiB/s] [=======================================================================================================================>] 100%
[ .... ] gcc-arm-8.3-2019.03-x86_64-arm-linux-gnueabihf.tar.xz:  244MiB [13.2MiB/s] [=======================================================================================================================>] 100%
[ .... ] downloading using torrent network [ gcc-arm-8.3-2019.03-x86_64-aarch64-linux-gnu.tar.xz ]
[ o.k. ] Verified [ MD5 ]                                                                                                                                                                                         
[ .... ] decompressing 
[ .... ] gcc-arm-8.3-2019.03-x86_64-aarch64-linux-gnu.tar.xz:  259MiB [12.9MiB/s] [=========================================================================================================================>] 100%
[ error ] ERROR in function source [ main.sh:272 ]
[ error ] Kernel branch not defined for this board [ default ]
[ o.k. ] Process terminated
```

Stadicus, 2019-11-20:
Trying to build the `next` branch on master branch works fine initially, but terminates later:

```bash
[ error ] ERROR in function compile_kernel [ compilation.sh:382 ]
[ error ] Kernel was not built [ @host ]
[ o.k. ] Process terminated
```

igorpecovnik, 2019-11-20:
BRANCH=default does not exists anymore. Working on proper document explaining this change based on https://github.com/armbian/build/pull/1586

Just change "default" to "legacy" and it should work.

Or use branch (LIB_TAG=v19.08 but not recommended since support will be dropped)

closed 2019-11-20 by igorpecovnik

Stadicus, 2019-11-20:
Thanks, will try that. 

What exactly does "support will be dropped" mean? Will LEGACY and CURRENT stay supported"?

igorpecovnik, 2019-11-20:
It's just a name change default = legacy, next = current and certain kernels has been dropped. Will be explained. I am working 24/7 do make this happen ...

Stadicus, 2019-11-20:
> I am working 24/7 do make this happen ...

Much appreciated, thanks!