armbian/build #1638: [BUG] build system do not include user configure file.

zhangn1985, 2019-11-27:
before today's update, I can create a config file in userpatch dir, `config-my.conf`

and I can use `./compile.sh docker my` to use veribles in config-my.conf.

after update, I cant.

zhangn1985, 2019-11-27:
https://github.com/armbian/build/commit/77a2dba1638d4d45a6ab2b2d5ab9505781378f07

this commit.

levindu, 2019-11-27:
Please test it with my newest pr rebase of docker-shell (link is above).

That should fix it. If not, please let me know. I'll keep a close eye on it.

zhangn1985, 2019-11-27:
@levindu I think you should properly fix this issue before submit new feature instead of a mixed commit with feature and fix.

zhangn1985, 2019-11-27:
could you simply  rename config-default.conf to config-XXX.conf, and add some varibles to config-XXX.conf, eg: 
```
SUBREVISION=.$(date +'%Y%m%d')
EXTRAWIFI=no
AUFS=no
RELEASE=buster
BRANCH=dev
EXPERT=yes
```
them build any device by`sudo ./compile.sh my` to check whether these varibles are used in this build.

zhangn1985, 2019-11-27:
https://github.com/armbian/build/blob/dac366f09fac8c4de4fddc0651564e3f1844308a/compile.sh#L172

I think a proper fix is to remove this line. because config-docker.conf already has a shift.

levindu, 2019-11-27:
https://github.com/armbian/build/pull/1612 should fix that, I double check:

```
./compile.sh docker my BRANCH=legacy
[ o.k. ] Using config file [ /home/dujw/p/armbian/docker-shell/config/templates/config-docker.conf ]
[ .... ] Using existed a armbian Docker container                               
[ o.k. ] Running the container             
[ o.k. ] Using config file [ /root/armbian/userpatches/config-my.conf ]
[ o.k. ] Command line: setting BRANCH to [ legacy ]
[ o.k. ] Preparing [ host ]
```

which introduces config-my.conf and combines with `K=V` in the command line.
Please test more.

commit de4cb772 (2019-11-28): Docker: add docker shell support (#1612)

This allows to enter the docker container in a shell by:

    ./compile.sh docker-shell
        BOARD=firefly-rk3399 BRANCH=dev RELEASE=bionic BUILD_DESKTOP=no

Then you can build the whole thing in the docker shell with:

    ./compile.sh

Once you need to build the U-Boot only for development purpose, you can run:

    # Optional: prepare the environment first if you had not run `./compile.sh`
    ./compile.sh 'prepare_host && compile_sunxi_tools && install_rkbin_tools'

    # build the U-Boot only
    ./compile.sh compile_uboot

If you prefer to use [... cut]
closed 2019-11-28 by igorpecovnik