armbian/build #1644: compile.sh ignores BOARD argument, shows interactive selection

Stadicus, 2019-11-28:
Build process no longer accepts the argument `BOARD` and always shows the interactive board selection.

Full command:
```
sudo ./compile.sh docker BOARD=rockpro64 KERNEL_ONLY=no KERNEL_CONFIGURE=no BUILD_MINIMAL=yes BUILD_DESKTOP=no RELEASE=bionic BRANCH=legacy WIREGUARD=no PROGRESS_LOG_TO_FILE=yes CLEAN_LEVEL=oldcache
```

CLI output (had to capture it with `tee`):
```
[ o.k. ] This script will try to update
Already up to date.
Your branch is up to date with 'origin/master'.
[ o.k. ] Using config file [ /mnt/hdd/github/bitbox-base/armbian/armbian-build/userpatches/config-docker.conf ]
[ .... ] Using existed a armbian Docker container
[ o.k. ] Running the container
[ o.k. ] Using config file [ /root/armbian/userpatches/config-example.conf ]
[ o.k. ] Command line: setting KERNEL_ONLY to [ no ]
[ o.k. ] Command line: setting KERNEL_CONFIGURE to [ no ]
[ o.k. ] Command line: setting BUILD_MINIMAL to [ yes ]
[ o.k. ] Command line: setting BUILD_DESKTOP to [ no ]
[ o.k. ] Command line: setting RELEASE to [ bionic ]
[ o.k. ] Command line: setting BRANCH to [ legacy ]
[ o.k. ] Command line: setting WIREGUARD to [ no ]
[ o.k. ] Command line: setting PROGRESS_LOG_TO_FILE to [ yes ]
[ o.k. ] Command line: setting CLEAN_LEVEL to [ oldcache ]

BOARD SELECTION DIALOG

[ o.k. ] Preparing [ host ]
[ o.k. ] Build host OS release [ bionic ]
[ o.k. ] Running in container [ docker ]
[ warn ] apt-cacher is disabled in containers, set NO_APT_CACHER=no to override
[ o.k. ] Checking for external GCC compilers
[ o.k. ] Downloading sources
[ o.k. ] Checking git sources [ u-boot-rockchip64 rockchip-master ]
[ .... ] Up to date
[ o.k. ] Checking git sources [ linux-rockchip64 4.4.197-1236-rockchip-ayufan ]
[ .... ] Up to date
[ o.k. ] Checking git sources [ arm-trusted-firmware-rockchip64 rockchip ]
[ .... ] Up to date
.
.
.
```

igorpecovnik, 2019-11-28:
Edit userpatches/config-docker.conf and remove shift in line 9

Stadicus, 2019-11-28:
Can confirm, that solved the issue for now. Thanks!

closed 2019-11-28 by Stadicus