armbian/build #1658: vagrant way of building

MarkoBursic, 2019-12-03:
I have followed the documentation [Vagrant tutorial](https://docs.armbian.com/Developer-Guide_Using-Vagrant/) So far I could install Vagrant and Ubuntu Bionic.

> # We have to be in the same directory as the Vagrant file.
> cd build
> 
> # And now we simply let vagrant create our box and bring it up. 
> vagrant up
> 
> # When the box has been installed we can get access via ssh.
> # (No need for passwords, Vagrant installs the keys we'll need.)
> vagrant ssh

It doesn't work for me. The build directory can't be seen once the VM starts. 
I did : "sudo su" ,  I can't find build directory, perhaps is somewhere, but where?

So, what I did was to clone the repository once again in the VM itself.
"root@ubuntu-bionic:/home/vagrant# git clone --depth 1 https://github.com/armbian/build"

Followed the tutorial, it's time to compile.
root@ubuntu-bionic:/home/vagrant/build# ./compile.sh

[ o.k. ] Using config file [ /home/vagrant/build/userpatches/config-example.conf ]
/home/vagrant/build/lib/main.sh: line 136: dialog: command not found
[ error ] ERROR in function source [ main.sh:138 ]
[ error ] No option selected
[ o.k. ] Process terminated

No fancy selection window is displayed, only errors. Where I did wrong?

martinayotte, 2019-12-03:
It is clearly missing the "dialog" tool.
You can install it using "apt-get install dialog" ...

igorpecovnik, 2019-12-03:
I think there are problems with Vagrant ... last time I was trying to get it working it was failing for me. It has to be reviewed by someone that uses this method or note that this method comes as is.

MarkoBursic, 2019-12-04:
This is my way, at least I do think it was this one:

vagrant plugin install vagrant-disksize
# then edit Vargantfile  and add:
config.disksize.size = '30GB' 
# otherwise the disk becomes full very soon

# Make the Vagrant box available. This might take a while but only needs to be done once.
vagrant box add ubuntu/bionic64

# Create the Vagrant environment
vagrant init ubuntu/bionic64

# If the box gets updated by the folks at HashiCorp, we'll want to update our copy too.
# This only needs done once and a while.
vagrant box update

# And now we simply let vagrant create our box and bring it up. 
vagrant up

# When the box has been installed we can get access via ssh.
# (No need for passwords, Vagrant installs the keys we'll need.)
vagrant ssh

# run update
apt-get update

# install dialog
apt-get install dialog

# Clone the project.
git clone --depth 1 https://github.com/armbian/build

cd build

sudo ./compile.sh

#now it takes hours
# type exit until you are out of ubuntu window
exit 

# copy the img to local disk
# export the ssh config with the private key
vagrant ssh-config > config.txt

# example of copying a created img file to my local dir
scp -F config.txt default:/home/vagrant/build/output/images/Armbian_19.11.3_Rockpi-4b_buster_current_5.4.1.img C:\Users\Marko\Downloads

# shut down VM
vagrant halt

# if and only if you won't need to build again
vagrant destroy

igorpecovnik, 2019-12-04:
But size is already defined big enough: https://github.com/armbian/build/blob/master/config/templates/Vagrantfile#L23

It must be some other problem?

belfastraven, 2019-12-08:
After the changes that moved the Vagrantfile to the Userpatches directory,  the documentation no longer matches what needs to happen.  The Userpatches directory seems not to be part of the initial git clone of build,  but seems to be built after running a compile.sh on the host.  At any rate,  if one Inits the vagrant box,  all that does is put a vanilla (non-armbian)  shell in the build directory.  The Vagrantfile in the Userpatches directory has all of the necessary armbian  configuration stuff, symllinks, etc in it,   so if one does a vagrant up from that directory, one has a properly configured machine.  Is there a possibility of moving thast Vagranty file  and the associated conf file to some directory that would be immediately available after the initial cloning of build?  They we would easily change the documentation to instruct people to run "vagrant up" from that directory.

igorpecovnik, 2019-12-08:
> They we would easily change the documentation to instruct people to run "vagrant up" from that directory.

Means instructing to: 

go to config/templates and run vagrant up from there? 

Easiest ... but it doesn't look like an initiative location where one would find this without looking to the docs, does it?

Any more convenient way like moving this file here and there, creating new folder containing stuff for virtual / containerized building?

belfastraven, 2019-12-08:
I believe most of the latest issues for people are caused by the documentation no longer matching what should happen,  but I guess we all  know that some people won't read doc no matter how good it is.  In that case, for Vagrant, at least.  it might be a good idea to have a folder named "vagrant"  in the build directory containing  the Vagrantfile and conf file,  and a "readme"  with a simple set up instructions. I haven't run the docker stuff lately,  so I don't kinow if having a "docker" folder would also be useful.  Perhaps the the vagrant option on ./compile.sh should be remove, also,  since it will only work for some linux distros, and not for Mac or windows.    I think the aim here is to make it easy for people to find the information and successfully run the build system so they don't have to bother you about it..Right? :-)

AS I mentioned in the forum a while ago,  I'm happy to rewrite documentation  or test.  I'm less comfortable trying to change the build system.

igorpecovnik, 2019-12-08:
> AS I mentioned in the forum a while ago, I'm happy to rewrite documentation or test. I'm less comfortable trying to change the build system.

Build system changes can be problematic even for most experienced / involved people. One must do lots of tests to make sure if some fix/feature didn't open another one.

Vagrant way of building is not that important but since we have it and it works - if properly setup - perhaps just update documentation to match reality should do. Thank you for helping out.

igorpecovnik, 2019-12-23:
https://github.com/armbian/documentation/commit/0f45431222702da10830a2d95b12dd67305532de

closed 2019-12-23 by igorpecovnik