armbian/build #1675: PPA bug for NanoPi-M4 and Tinkerboard

dragonlost, 2019-12-14:
I tested these version: 
for nanopi :
- Armbian_5.95_Nanopim4_Ubuntu_bionic_default_4.4.190_desktop --> no ppa bug
- Armbian_5.98_Nanopim4_Ubuntu_bionic_default_4.4.192_desktop --> ppa bug
- Armbian_19.11.3_Nanopim4_bionic_legacy_4.4.192_desktop --> ppa bug

for tinkerboard : 
- Armbian_5.90_Tinkerboard_Ubuntu_bionic_default_4.4.182_desktop --> no ppa bug
- Armbian_5.95_Tinkerboard_Ubuntu_bionic_default_4.4.190_desktop --> ppa bug
- Armbian_19.11.3.346_Tinkerboard_bionic_legacy_4.4.199_desktop --> need test

The bug is as follows. When I start the armbian image without any modification and that I add a ppa then that if, is added as "focal"(20.04LTS) instead of "bionic" (18.04LTS).

dragonlost, 2019-12-14:
armbian 5.90 is last free bug version (for nanopi-m4 and tinkerboard)

igorpecovnik, 2019-12-14:
This bug has nothing to do with Armbian.

igorpecovnik, 2019-12-14:
Try this on non working ones: apt install lsb-release , remove ppa and add again.

dragonlost, 2019-12-14:
I don't agree it's an armbian bug.

Whatever the PPA added the same bug appears for versions of Armbian higher than 5.90 on Nanopi_m4 and tinkerboard.

Several people in my community (NAFABox) report the same bug

dragonlost, 2019-12-14:
I will test lsb-release

igorpecovnik, 2019-12-14:
Armbian is build in deterministic way ... but if there are bugs or changes in upstream packages, there is little we can do about. 

Armbian_5.90_Nanopim4_Ubuntu_bionic_default_4.4.179_desktop --> no ppa bug
Armbian_5.95_Nanopim4_Ubuntu_bionic_default_4.4.190_desktop --> ppa bug

Have you run apt update and upgrade on both before adding PPA?

Try to figure out how this PPA is determining which release is on. This seems to be the center of this problem.

dragonlost, 2019-12-14:
no I did not update with apt before adding a ppa.

this bug is not for one PPA is for all PPA add by : `sudo apt-add-repository ppa:...`

igorpecovnik, 2019-12-14:
> no I did not update with apt before adding a ppa.

OK, do that.

I suspect something ... will dig into this when possible.

igorpecovnik, 2019-12-14:
Run few Python commands on the broken/good one:
>>> import platform
>>> platform.dist()

You should get something like this:

('debian', '10.2', '')

dragonlost, 2019-12-14:
ok `lsb-release` is not possible because is ubuntu system.

the return of `platform.dist()` --> ('Ubuntu', '18.04', 'bionic')
test on Armbian_19.11.3_Nanopim4_bionic_legacy_4.4.192_desktop
My process : 
- Flash new sd card with Armbian_19.11.3_Nanopim4_bionic_legacy_4.4.192_desktop with BalenaEtcher
- Run on my NanoPi-M4
- Create new user : `nafa`
- right click : `open terminal here`
- `sudo apt-add-repository ppa:...`
- `cat /etc/apt/sources.list.d/... .list`
--> deb http://ppa.launchpad.net/... __focal__ main

No update/upgrade

dragonlost, 2019-12-14:
ok is `lsb_release -a` --> 

Distributor ID: Ubuntu
Description: Ubuntu 18.04.3 LTS
Release: 18.04
Codename: bionic

igorpecovnik, 2019-12-14:
This apt-add-repository is using those Python libraries to determine OS version and if they report properly, while script fails ... strange. 

lsb_release is our script while original one is installed from package lsb-release (- not _ ). I am out of office so can't test and dig more. Later.

dragonlost, 2019-12-14:
yes is very strange. Thanks for your help

igorpecovnik, 2019-12-14:
Related: https://unix.stackexchange.com/questions/284613/add-apt-repository-sets-incorrect-distribution-name

What do you have here: /etc/lsb-release

dragonlost, 2019-12-14:
`cat /etc/ls-release` -->
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=18.04
DISTRIB_CODENAME=bionic
DISTRIB_DESCRIPTION="Ubuntu 18.04.3 LTS"

igorpecovnik, 2019-12-15:
Can replicate but can't find where is the problem yet. This warning is the closest I got: 
_main__:1: DeprecationWarning: dist() and linux_distribution() functions are deprecated in Python 3.5

dragonlost, 2019-12-15:
> 
> 
> Can replicate but can't find where is the problem yet. This warning is the closest I got:
> _main__:1: DeprecationWarning: dist() and linux_distribution() functions are deprecated in Python 3.5

yes dist() is deprecated on python 3. but is work on python 2. I have test this command on python 2.
need use platform.linux_distribution() on python 3

igorpecovnik, 2019-12-15:
Found the source of the problem. Our bash replacement script seems to not work best: https://github.com/armbian/build/blob/master/packages/bsp/common/usr/bin/lsb_release Will try to implement a fix tomorrow.

dragonlost, 2019-12-16:
Great. Thank you very much for all your work!

closed 2019-12-16 by igorpecovnik