armbian/build #1681: ODROID HC2 bricked after kernel update from this evening

ThomasKaiser, 2019-12-17:
Downloaded `Armbian_19.11.3_Odroidxu4_buster_legacy_4.14.150.7z` few hours ago, flashed to SD card, booted ODROID HC2, applied all updates (kernel update included), spent some hours on configuration and then got notified of 2 new Armbian updates half an hour ago:

    root@brique-muc:/home/tk# apt upgrade
    Reading package lists... Done
    Building dependency tree       
    Reading state information... Done
    Calculating upgrade... Done
    The following packages will be upgraded:
      linux-dtb-legacy-odroidxu4 linux-image-legacy-odroidxu4
    2 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
    Need to get 16.1 MB of archives.
    After this operation, 2048 B of additional disk space will be used.
    Do you want to continue? [Y/n] 
    Get:1 https://apt.armbian.com buster/main armhf linux-dtb-legacy-odroidxu4 armhf 19.11.4 [80.5 kB]
    Get:2 https://apt.armbian.com buster/main armhf linux-image-legacy-odroidxu4 armhf 19.11.4 [16.0 MB]
    Fetched 16.1 MB in 2s (7177 kB/s)                       
    (Reading database ... 52281 files and directories currently installed.)
    Preparing to unpack .../linux-dtb-legacy-odroidxu4_19.11.4_armhf.deb ...
    Unpacking linux-dtb-legacy-odroidxu4 (19.11.4) over (19.11.3.20191205) ...
    Preparing to unpack .../linux-image-legacy-odroidxu4_19.11.4_armhf.deb ...
    Unpacking linux-image-legacy-odroidxu4 (19.11.4) over (19.11.3.20191205) ...
    Setting up linux-dtb-legacy-odroidxu4 (19.11.4) ...
    Setting up linux-image-legacy-odroidxu4 (19.11.4) ...
    update-initramfs: Generating /boot/initrd.img-4.14.157-odroidxu4
    update-initramfs: Converting to u-boot format
    
    root@brique-muc:/home/tk# reboot
    Connection to brique-muc closed by remote host.
    Connection to brique-muc closed.
    
    mac-tk:~ tk$ sleep 600 ; ping brique-muc
    PING brique-muc.fritz.box (192.168.83.183): 56 data bytes
    Request timeout for icmp_seq 0
    Request timeout for icmp_seq 1
    Request timeout for icmp_seq 2
    Request timeout for icmp_seq 3
    ^C
    --- brique-muc.fritz.box ping statistics ---
    5 packets transmitted, 0 packets received, 100.0% packet loss
    mac-tk:~ tk$ 

When unplugging power from the board and reinserting the power jack the following happens (no serial console available). Red and green led light up immediately and the blue led follows within 1.5 seconds and remains solid blue forever (never starts with heartbeat blinking).

Is this a know issue?

igorpecovnik, 2019-12-17:
Tested upgrade: https://forum.armbian.com/topic/12437-enable-pids-cgroup/?tab=comments#comment-91184

igorpecovnik, 2019-12-17:
Another test by me http://ix.io/24JR ... without professional grade testing is impossible to catch such things.

belegdol, 2019-12-18:
something fishy is going on definitely. What I suspected to be intermittent network issue was actually a real problem.
My HC1 booted after I connected the UART cable to it. As I am using the plastic cover, this necessitated a several seconds long power disconnection.

igorpecovnik, 2019-12-18:
Update rolled back.

ThomasKaiser, 2020-01-03:
> Update rolled back.

But my system is still bricked. I haven't opted for volunteer testing of some unstable/experimental kernel branch. I simply made (again) the mistake to trust into the 'stable' Armbian repo and got a bricked installation (maybe the 10th in three years now). Why has this to happen constantly?

closed 2020-01-04 by igorpecovnik

igorpecovnik, 2020-01-04:
> I haven't opted for volunteer testing

It's obvious what went wrong once solved, but prior to that ...

Armbian don't have expensive automated testing facility - and until amateur version is not built and put into action, a volunteer testing will still be needed. This particular bug was almost impossible to track down with a quick 1-2 boots / upgrades which we can afford to do. On top of that, image worked fine, once / if booted ... 

Professional dev teams have not just one but teams developing and doing just testings. We have this quickly put together, WIP / early development for now:
https://github.com/armbian/autotests
... but if there will be no input from others, others have nothing to complain about.

Images and update will be pushed later after running one more test cycle. Just to make sure.

I had to waste a day of my holiday time to solve this. Cheers!

igorpecovnik, 2020-01-04:
Repository updated.