armbian/build #1684: Building package for linux headers

mmriech, 2019-12-19:
As I'm currently trying to understand why DKMS does not work with recent kernels I had a look on the creation of the package for the linux headers. Some issues came up I would like to discuss.

1. First of all a "concrete" finding: The postinst-script. The builddeb-script of the kernel is patched in a way that is automatically creates a postinst-script that is executed after the header-package is installed One of its task is displaying "Compiling headers - please wait ..." but there is also some work done in the background. Unfortunately the output of the commands that are executed are discarded: `>/dev/null 2>&1` This is useful to reduce noise, but it completely hides error output. Is this indended?

2. And then a more general finding/question: The header package does not have any dependencies. This is quite problematic when installing the headers on a minimal system. Installing the headers requires make. If it is not available the postinst-script cannot be executed successfully. But unfortunately due to issue (1) this is hidden. And make is not enough. Recent kernels also require tools like flex, bison and libssl-dev. They are all not automatically installed by the kernel headers due to the missing dependencies. Debian's own kernel headers, like for amd64, have quite a lot of dependencies. I assume they cover all tools required to use the kernel headers. Shouldn't Armbian do it the same way or has there been a decision to handle the dependencies somehow else?

igorpecovnik, 2019-12-19:
> Shouldn't Armbian do it the same way or has there been a decision to handle the dependencies somehow else?

There was not special decisions ... more like a natural development + avoiding to deal with scattered patches on a kernel which are going to be dropped anyway ... and create some better solution. 

Our challenges: 10+ different kernel sources, many of them from dirty development drops, up to two arch. Each needed some adjustments to provide the same packaging structure ... Debian packaging way was just a template and 10y old is very different than the actual one.

Today we have only a few patches https://github.com/armbian/build/tree/master/patch/misc and much less kernel sources (we recently dropped all old junk) which means its much simpler to get a picture what can be done.  I already made a plan to do something about to make this part even better [AR-42]

Once we only have one script to maintain ... its much easier to address problems you are bringing up - solve dependencies and logging properly ...

mmriech, 2019-12-27:
Closing this issue. Pull request #1679 is going to add the dependencies to the currently supported kernels.

count-doku, 2019-12-28:
Is this and the PR related to #1369 ?
Seems like the same to me.

igorpecovnik, 2019-12-28:
Yes, its the same.

closed 2019-12-28 by mmriech