armbian/build #1687: docker build fails when adding ubuntu key

nickbreen, 2019-12-22:
This is a clean checkout from git:

```bash
nick@skull:~/src/armbian/build$ git remote -v
origin	https://github.com/armbian/build (fetch)
origin	https://github.com/armbian/build (push)

nick@skull:~/src/armbian/build$ git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
```

Recent version of docker:

```bash
nick@skull:~/src/armbian/build$ docker --version
Docker version 18.09.9, build 1752eb3
```

```bash
nick@skull:~/src/armbian/build$ ./compile.sh docker KERNEL_ONLY=yes BOARD=pinebook-a64 BRANCH=next KERNEL_CONFIGURE=yes
[ o.k. ] This script will try to update
Already up to date.
Already on 'master'
Your branch is up to date with 'origin/master'.
[ o.k. ] Using config file [ /home/nick/src/armbian/build/userpatches/config-docker.conf ]
[ .... ] Docker container not found or out of date 
[ .... ] Building a Docker container 
Sending build context to Docker daemon  16.38kB
Step 1/10 : FROM ubuntu:18.04
 ---> 549b9b86cb8d
Step 2/10 : RUN apt update && DEBIAN_FRONTEND=noninteractive apt install -y gnupg gnupg1 gpgv1 --no-install-recommends
 ---> Using cache
 ---> 5ff28310dafb
Step 3/10 : RUN sh -c  "if [ x"" != x$http_proxy ]; then                 apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --keyserver-options http-proxy=$http_proxy --recv-keys ED75B5A4483DA07C >/dev/null 2>&1; 		    else 			    apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys ED75B5A4483DA07C >/dev/null 2>&1; 		    fi"
 ---> Running in 78ce9c44d660
The command '/bin/sh -c sh -c  "if [ x"" != x$http_proxy ]; then                 apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --keyserver-options http-proxy=$http_proxy --recv-keys ED75B5A4483DA07C >/dev/null 2>&1; 		    else 			    apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys ED75B5A4483DA07C >/dev/null 2>&1; 		    fi"' returned a non-zero code: 2

```

nickbreen, 2019-12-23:
The internal error is:

```
gpg: keyserver receive failed: No keyserver available
```

When running:

```
apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys ED75B5A4483DA07C
```

igorpecovnik, 2019-12-23:
> The internal error is:

We would need to rely on more than one server to overcome this problem.

nickbreen, 2019-12-23:
Bah!

My docker daemon was configured with a nameserver IP address that no longer worked. Despite it having a good nameserver configured too.

closed 2019-12-23 by nickbreen