armbian/build #1728: Error while trying to compile Armbian

juanesf, 2020-01-09:
I have followed the common instructions:
apt -y install git
git clone https://github.com/armbian/build
cd build
./compile.sh

but when executing the script the following occurs:
[ o.k. ] Using config file [ /home/usuario/build/userpatches/config-example.conf ]
/home/usuario/build/lib/main.sh: línea 136: dialog: orden no encontrada
[ error ] ERROR in function source [ main.sh:138 ]
[ error ] No option selected 
[ o.k. ] Process terminated

igorpecovnik, 2020-01-09:
Just try once again.

martinayotte, 2020-01-09:
"dialog: orden no encontrada" == "command not found"
do "apt-get install dialog"

juanesf, 2020-01-09:
"apt-get install dialog"

This has corrected the problem. Thank you

closed 2020-01-09 by juanesf

igorpecovnik, 2020-01-09:
But dialog should be installed https://github.com/armbian/build/blob/master/lib/general.sh#L703 which probably means our detection of apt background activities does not work properly and install fails ... at next start it works.