armbian/build #1733: Redis configuration

Averell7, 2020-01-11:
I just discovered that, when starting, redis issues 3 Warnings :

WARNING: The TCP backlog setting of 511 cannot be enforced because /proc/sys/net/core/somaxconn is set to the lower value of 128.

WARNING overcommit_memory is set to 0! Background save may fail under low memory condition. To fix this issue add 'vm.overcommit_memory = 1' to /etc/sysctl.conf and then reboot or run the command 'sysctl vm.overcommit_memory=1' for this to take effect.

WARNING you have Transparent Huge Pages (THP) support enabled in your kernel. This will create latency and memory usage issues with Redis. To fix this issue run the command 'echo never > /sys/kernel/mm/transparent_hugepage/enabled' as root, and add it to your /etc/rc.local in order to retain the setting after a reboot. Redis must be restarted after THP is disabled.

1. The first one is probably not a problem, but nevertheless the incoherence could (should ?) be fixed. 
2. The second one is more serious : Armbian may run on SBC with low memory. I guess few people will fill redis with tons of data, but nevertheless, fixing this should do no harm ?
3. The third one is more serious, but I don't know the advantages of THP.

aadityabagga, 2020-01-15:
Hi,

The warnings reported by redis are not specific to Armbian. Even when running Ubuntu on x86 server I get these warnings.

Can make the changes suggested by redis on a personal system, for widespread default more investigation can be done and reported to upstream Debian. My humble opinion.

Averell7, 2020-01-15:
OK, you are right, thank you. It was just an information.

closed 2020-01-30 by lanefu