armbian/build #1735: Make nand-sata-install options callable directly from commandline

friesenkiwi, 2020-01-12:
Hi,

I would like to invoke the options of the `nand-sata-install` script directly from commandline, without needing to go through the GUI selection menu. My use case is, I'm writing an ansible role for the helios4 and include an option to move the bootloader to nand. Currently, I need to pause the ansible execution, log in, execute the command, select the option 7, resume ansible execution.

So what I'm looking for, is something like `nand-sata-install --option 7`. Is this maybe already doable or could that be included as a new feature into the script? (I tried https://docs.ansible.com/ansible/latest/modules/expect_module.html but to no avail)

Thanks in advance for considering.

igorpecovnik, 2020-01-12:
> Thanks in advance for considering.

Wishes go here:
https://forum.armbian.com/forum/38-feature-requests/

This is amateur based project. If you want new features - this is the way:
https://www.armbian.com/get-involved
Well, there is another. If you find and pay someone to do that. Them remember to maintain it, otherwise it will be deleted at first sign of trouble.

count-doku, 2020-01-12:
I would also suggest posting this on the forum. It sounds helpful to have for automated installs and after a quick look at nand-sata-install seems not to difficult. 

@friesenkiwi Did you try to comment out all dialog options and add functionality to use a command line argument to just to the right case directly?

igorpecovnik, 2020-01-12:
> It sounds helpful to have for automated installs

I agree it would be nice to have. Also for armbian-config. But its some work and it also has to be tested, documented and maintained ...

psztoch, 2021-05-10:
I have already failed after two attempts to refactor nand-sata-install.
Unfortunately, the number of variants (sometimes contradictory) is terrifying.
And each case has a clear sense and probably the users who use it.
I miss the most flexible and more extensive partitioning (/boot, /, /var, swap ...) and full migration to NVME (with /boot too).

Finally, I made my script for a specific application and it came out 50 times shorter and easy to manage.

pedralmeida22, 2024-10-15:
Any new related to adding this feature?

igorpecovnik, 2025-01-19:
If anyone wants to proceed, this way https://github.com/armbian/configng is better.

- good support for testing
- parameters support
- modular structure
- (unit) testing

Perhaps this way:

- separate partitioning
- functions per each install case

Closing - we will only patch the tool that sits in this code. If we weren't able to develop it from scratch in five years, it probably means we are far away :( At least regulars that are looking on those issues.

closed 2025-01-19 by igorpecovnik