armbian/build #1742: Broken thinkerboard bionic (current) build

Zatexev, 2020-01-16:
I was doing buster and bionic builds, withtou problem using the kernel 5.4.10 or 5.4.11, but after the update to 5.4.12 its not working any build anymore, yes I already cleanup and git clone again.

I am using ./compile.sh docker
both variantes not working anymore
###
BOARD=tinkerboard
BRANCH=current
RELEASE=buster
BUILD_MINIMAL=yes 
BUILD_DESKTOP=no
###
or the bionic buld with desktop
BOARD=tinkerboard
BRANCH=current
RELEASE=bionic
BUILD_MINIMAL=no
BUILD_DESKTOP=YES

`Displaying message: Using user configuration override userpatches/lib.config info

## BUILD SCRIPT ENVIRONMENT

Repository: 
Version: 

Host OS: bionic
Host arch: amd64
Host system: Linux 2f92539c01c1 4.18.0-151.el8.x86_64 #1 SMP Wed Dec 4 17:04:30 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
Virtualization type: docker

## Build script directories
Build directory is located on:
TARGET        SOURCE                                                                    FSTYPE  AVAIL
/root/armbian /dev/mapper/cs_ftsbuilder02-root[/opt/tinkerboard/asus-tinkerboard-build] xfs    206.9G

Build directory permissions:
# file: /root/armbian
# owner: root
# group: root
user::rwx
group::r-x
other::r-x

Temp directory permissions:
# file: /root/armbian/.tmp
# owner: root
# group: root
user::rwx
group::rwx
other::r-x

## BUILD CONFIGURATION

Build target:
Board: tinkerboard
Branch: current
Minimal: yes
Desktop: no

Kernel configuration:
Repository: git://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git
Branch: branch:linux-5.4.y
Config file: linux-rockchip-current

U-boot configuration:
Repository: git://git.denx.de/u-boot.git
Branch: tag:v2018.11
Config file: tinker-rk3288_defconfig

Partitioning configuration:
[... 96 lines cut ...]
Displaying message: * [\e[32ml\e[0m][\e[32mc\e[0m] 2004-Remove-the-dependency-to-the-clk_mali-symbol.patch  info
Displaying message: * [\e[32ml\e[0m][\e[32mc\e[0m] 2005-drivers-mmc-dw-mci-rockchip-Handle-ASUS-Tinkerboard.patch failed wrn
Displaying message: * [\e[32ml\e[0m][\e[32mc\e[0m] 2006-soc-rockchip-power-domain-export-idle-request.patch failed wrn
Displaying message: * [\e[32ml\e[0m][\e[32mc\e[0m] 2007-drivers-wifi-ath9k-reverse-do-not-use-bulk-on-EP3-and-EP4.patch failed wrn
Displaying message: * [\e[32ml\e[0m][\e[32mc\e[0m] 2008-clk-rockchip-rk3288-Support-for-dedicating-NPLL-to-a.patch failed wrn
Displaying message: * [\e[32ml\e[0m][\e[32mc\e[0m] 2009-drm-dw_hdmi-rockchip-better-clock-selection-logic-an.patch failed wrn
Displaying message: * [\e[32ml\e[0m][\e[32mc\e[0m] 2013-spi-Added-support-for-Tinkerboard-s-SPI-interface.patch failed wrn
Displaying message: * [\e[32ml\e[0m][\e[32mc\e[0m] 261_gpiomem_driver.patch failed wrn
Displaying message: * [\e[32ml\e[0m][\e[32mc\e[0m] 3000_DTS_rearrange_thermal_zones.patch failed wrn
Displaying message: * [\e[32ml\e[0m][\e[32mc\e[0m] 3001_Tinkerboard-4.19-audio.patch failed wrn
Displaying message: * [\e[32ml\e[0m][\e[32mc\e[0m] board-tinkerboard-enable-emmc-support.patch  info
Displaying message: * [\e[32ml\e[0m][\e[32mc\e[0m] brcmfmac-add-ap6330-firmware.patch failed wrn
Displaying message: * [\e[32ml\e[0m][\e[32mc\e[0m] general-adjust-tinker-dts-hdmi-sound.patch failed wrn
Displaying message: * [\e[32ml\e[0m][\e[32mc\e[0m] general-fix-reboot-from-kwiboo.patch failed wrn
Displaying message: * [\e[32ml\e[0m][\e[32mc\e[0m] remove-broken-dtb.patch failed wrn
Displaying message: * [\e[32ml\e[0m][\e[32mc\e[0m] xt-q8l-v10-add-device-tree.patch  info
Displaying message: * [\e[32ml\e[0m][\e[32mc\e[0m] xt-q8l-v10-add-dts-makefile.patch failed wrn
Displaying message: * [\e[32ml\e[0m][\e[32mc\e[0m] xt-q8l-v10-remote-keymap.patch failed wrn
Displaying message: Compressing sources for the linux-source package
Displaying message: Compiling current kernel 5.4.12 info
Displaying message: Compiler version arm-linux-gnueabihf-gcc 8.3.0 info
Displaying message: Using kernel config file config/kernel/linux-rockchip-current.config info
Displaying message: ERROR in function compile_kernel compilation.sh:378 err
Displaying message: Kernel was not built @host err
Displaying message: Process terminated  info`

martinayotte, 2020-01-16:
For more detailed errors, look at output/debug/compilation.log
I bet you don't have "dialog" tool on you machine ...
To install it, do "apt-get install dialog" !

Zatexev, 2020-01-16:
I am using docker as builder... compilation.log shows
2 errors:
make[1]: *** [fs/read_write.o] Error 1
make: *** [fs] Error 2

full compilation.log
`
	== u-boot ==

.config:1418:warning: override: reassigning to symbol BOOTDELAY
common/autoboot.c: In function ‘autoboot_command’:
common/autoboot.c:348:4: warning: implicit declaration of function ‘rk3288_maskrom_ctrl’ [-Wimplicit-function-declaration]
    rk3288_maskrom_ctrl(false);
    ^~~~~~~~~~~~~~~~~~~
common/autoboot.c:349:4: warning: implicit declaration of function ‘usb_current_limit_ctrl’ [-Wimplicit-function-declaration]
    usb_current_limit_ctrl(false);
    ^~~~~~~~~~~~~~~~~~~~~~

	== kernel ==

drivers/gpu/arm/midgard/mali_kbase_jd_debugfs.c: In function ‘kbase_jd_debugfs_fence_info’:
drivers/gpu/arm/midgard/mali_kbase_jd_debugfs.c:43:6: warning: this statement may fall through [-Wimplicit-fallthrough=]
   if (0 == res) {
      ^
drivers/gpu/arm/midgard/mali_kbase_jd_debugfs.c:48:2: note: here
  case BASE_JD_REQ_SOFT_FENCE_WAIT:
  ^~~~
drivers/gpu/arm/midgard/mali_kbase_sync_file.c: In function ‘kbase_sync_fence_info_get’:
drivers/gpu/arm/midgard/mali_kbase_sync_file.c:300:51: warning: format ‘%u’ expects argument of type ‘unsigned int’, but argument 5 has type ‘u64’ {aka ‘long long unsigned int’} [-Wformat=]
  scnprintf(info->name, sizeof(info->name), "%llu#%u",
                                                  ~^
                                                  %llu
     fence->context, fence->seqno);
                     ~~~~~~~~~~~~                   
drivers/gpu/arm/midgard/mali_kbase_fence.c: In function ‘kbase_fence_fence_value_str’:
drivers/gpu/arm/midgard/mali_kbase_fence.c:65:24: warning: format ‘%u’ expects argument of type ‘unsigned int’, but argument 4 has type ‘u64’ {aka ‘long long unsigned int’} [-Wformat=]
  snprintf(str, size, "%u", fence->seqno);
                       ~^   ~~~~~~~~~~~~
                       %llu
fs/read_write.c:526:13: error: redefinition of ‘vfs_readf’
 vfs_readf_t vfs_readf(struct file *file)
             ^~~~~~~~~
fs/read_write.c:502:13: note: previous definition of ‘vfs_readf’ was here
 vfs_readf_t vfs_readf(struct file *file)
             ^~~~~~~~~
fs/read_write.c:537:14: error: redefinition of ‘vfs_writef’
 vfs_writef_t vfs_writef(struct file *file)
              ^~~~~~~~~~
fs/read_write.c:514:14: note: previous definition of ‘vfs_writef’ was here
 vfs_writef_t vfs_writef(struct file *file)
              ^~~~~~~~~~
fs/read_write.c:548:13: error: redefinition of ‘vfs_readf’
 vfs_readf_t vfs_readf(struct file *file)
             ^~~~~~~~~
fs/read_write.c:502:13: note: previous definition of ‘vfs_readf’ was here
 vfs_readf_t vfs_readf(struct file *file)
             ^~~~~~~~~
fs/read_write.c:559:14: error: redefinition of ‘vfs_writef’
 vfs_writef_t vfs_writef(struct file *file)
              ^~~~~~~~~~
fs/read_write.c:514:14: note: previous definition of ‘vfs_writef’ was here
 vfs_writef_t vfs_writef(struct file *file)
              ^~~~~~~~~~
fs/read_write.c:570:13: error: redefinition of ‘vfs_readf’
 vfs_readf_t vfs_readf(struct file *file)
             ^~~~~~~~~
fs/read_write.c:502:13: note: previous definition of ‘vfs_readf’ was here
 vfs_readf_t vfs_readf(struct file *file)
             ^~~~~~~~~
fs/read_write.c:581:14: error: redefinition of ‘vfs_writef’
[... 1936 lines cut ...]
 #define IEEE80211_MAX_AMPDU_BUF 0x40
 
In file included from drivers/net/wireless/rtl8811cu/include/osdep_service_linux.h:83,
                 from drivers/net/wireless/rtl8811cu/include/osdep_service.h:50,
                 from drivers/net/wireless/rtl8811cu/include/drv_types.h:27,
                 from drivers/net/wireless/rtl8811cu/hal/btc/mp_precomp.h:18,
                 from drivers/net/wireless/rtl8811cu/hal/btc/halbtc8821cwifionly.c:15:
./include/linux/ieee80211.h:1451: note: this is the location of the previous definition
 #define IEEE80211_MAX_AMPDU_BUF  0x100
 
In file included from drivers/net/wireless/rtl8811cu/include/drv_types.h:30,
                 from drivers/net/wireless/rtl8811cu/hal/btc/mp_precomp.h:18,
                 from drivers/net/wireless/rtl8811cu/hal/btc/halbtc8821c2ant.c:16:
drivers/net/wireless/rtl8811cu/include/wifi.h:1032: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 #define IEEE80211_MAX_AMPDU_BUF 0x40
 
In file included from drivers/net/wireless/rtl8811cu/include/osdep_service_linux.h:83,
                 from drivers/net/wireless/rtl8811cu/include/osdep_service.h:50,
                 from drivers/net/wireless/rtl8811cu/include/drv_types.h:27,
                 from drivers/net/wireless/rtl8811cu/hal/btc/mp_precomp.h:18,
                 from drivers/net/wireless/rtl8811cu/hal/btc/halbtc8821c2ant.c:16:
./include/linux/ieee80211.h:1451: note: this is the location of the previous definition
 #define IEEE80211_MAX_AMPDU_BUF  0x100
 
`

igorpecovnik, 2020-01-16:
One of the patches got in the conflict with mainstream code. I would say to remove mali patch and try again. I will look into this later.

igorpecovnik, 2020-01-16:
I can't recreate on a clean native nor in a Docker environment. I would say starts from a clean Docker image.

closed 2020-01-16 by igorpecovnik