armbian/build #1744: Low(er) memory and overall performance with rockchip64/current

ThomasKaiser, 2020-01-17:
Just FYI: Running sbc-bench on Renegade with 5.4 kernel results in abysmal memory performance which also affects the overall performance.

[Renegade at 1.5GHz with 5.4.8-rockchip64](http://ix.io/27Dq):

    standard memcpy                                      :    878.6 MB/s (1.7%)
    standard memset                                      :   3808.1 MB/s (2.7%)
    7-zip total: 3074,3079,3050
    cpuminer: 4.41,4.38,4.37,4.36,4.35,4.34,4.33,4.32,4.31,4.30,4.29,4.26

[Rock64 at just 1.3 GHz with 4.18.0-rc5-1050-ayufan-ge70bd2ab8802](http://ix.io/1iH4):

    standard memcpy                                      :   1338.6 MB/s (0.5%)
    standard memset                                      :   5766.7 MB/s
    7-zip total: 3533,3532,3539
    cpuminer: 4.69,4.68,4.67,4.66,4.65,4.64,4.63,4.62

Please keep in mind that the Rock64 numbers were made with just 1.3 GHz and GCC 7.3 vs. 1.5 GHz and GCC 8.3.

I can't test with Rock64/current since owning only a pre-production board that doesn't boot or initialize Ethernet (different PHY) with mainline kernel. And I have no idea whether the culprit is with kernel, bootloader or whatever...

BTW: `armbianmonitor -z` functionality has been invented also for this reason: (automatic) checking whether an upgrade breaks stuff (performance in this case -- who wants a newer release that performs worse?)

igorpecovnik, 2020-01-17:
Interesting findings.

> functionality has been invented also for this reason: (automatic) checking whether an upgrade breaks stuff

The tool do the right job, but what we don't have is automated data collection and analysis. To put out a red flag (only) on a measurement that falls out of tolerance. That differs more then x % from previous ones. I hope we will be able to setup that once.

I think also USB3.0 is not working full speed in current yet, certainly some other minor things, but general ones are fixed which is the reason current exists while (stalled) DEV remains attached to Ayufan's.

Tonymac32, 2020-01-17:
Renegade is DDR4, so this should be fun.  Just starting my day, so we'll
see if I can check this later.  I also have a newer Rock64.

TRSx80, 2020-01-30:
> automated data collection and analysis

I know this is probably best discussed elsewhere, but for the love of $diety can we please make this opt-in only (for ex. as Debian proper does with popcon)?

ThomasKaiser, 2020-01-30:
> can we please make this opt-in only

Igor was talking about some (never happening) automated testing prior to releasing a new version by 'board maintainers' (or whatever group of people currently is lacking).

So nothing new on topic...

TRSx80, 2020-01-30:
Yeah sorry for bumping your post. I ended up lodging my concern [in the correct place](https://github.com/armbian/autotests/issues/1) which is what I should have done in the first place.

It's always good to see you around TK, even if just reporting bugs. I learned a lot from all your very informative posts over the years, I'm sure many others as well. Cheers.

lanefu, 2020-08-07:
Well here's armbian kernel 5.7.12 on my rock64 that doesn't have a heat sink

```
memcpy: 1409.1 MB/s
memset: 5820.4 MB/s
```

Manouchehri, 2022-10-20:
Even with a ton of hardening flags enabled, my results on a NanoPi R4S[e] are much better.

https://gist.github.com/Manouchehri/747bf05aa0794fc8a550a535381c1797

```
standard memcpy                                      :   3542.4 MB/s
standard memset                                      :   8594.3 MB/s (0.7%)
```

Tonymac32, 2023-10-03:
@lanefu @armbian/boards-rockchip  care to post current output of LTS and edge kernels to see where they stand?  Then possibly close this if it's been rectified.

Tonymac32, 2023-11-13:
Kernel 6.6, just did the test, 1.3 GHz:

memcpy: 1596.1 MB/s
mwmset: 6035.2 MB/s

closed 2023-11-13 by Tonymac32