armbian/build #1751: Installing kernel-headers 4.4.210-rockchip64 "fails"

mmriech, 2020-01-19:
When building kernel-headers 4.4.210 for rockchip64 the installation "fails" reporting the following error

`make: *** No rule to make target 'prepare0', needed by 'vdso_prepare'.  Stop.`

I put it in quotation marks because only internal steps fail but the postinst succeeds (reason is explained here: https://github.com/armbian/build/issues/830#issuecomment-575987745).

I propose two measures:
* Fix the problem in kernel 4.4.210 like proposed here: https://patchwork.kernel.org/patch/10661821/. I already tested that and it seems to work.
* Allow postinst to fail if an internal command fails. Details are explained in https://github.com/armbian/build/issues/830#issuecomment-575987745. I hope that measure is sufficient, I did not test it yet.

If there are no strong opinions against that procedure I can create a PR for that.

igorpecovnik, 2020-01-19:
> I already tested that and it seems to work.

I know it didn't work before on 4.4.y and if it works, do send a PR.

mmriech, 2020-01-22:
>     * Fix the problem in kernel 4.4.210 like proposed here: https://patchwork.kernel.org/patch/10661821/. I already tested that and it seems to work.
> 
This has been fixed by PR #1753 

>     * Allow postinst to fail if an internal command fails. Details are explained in [#830 (comment)](https://github.com/armbian/build/issues/830#issuecomment-575987745). I hope that measure is sufficient, I did not test it yet.
> 
This is still in discussion in PR #1754

mmriech, 2020-02-09:
> 
> > ```
> > * Allow postinst to fail if an internal command fails. Details are explained in [#830 (comment)](https://github.com/armbian/build/issues/830#issuecomment-575987745). I hope that measure is sufficient, I did not test it yet.
> > ```
> 
> This is still in discussion in PR #1754

PR #1754 has been merged, so the postinst now fails if an internal command fails. So this issue has been solved now.

closed 2020-02-09 by mmriech