armbian/build #1769: wrong initramfs is created

NickeZ, 2020-01-29:
Hey,

When I build for the rockpro64 I get the following content in the boot directory:

```
drwxr-xr-x  3 root root     4096 Jan 29 00:44 .
drwxr-xr-x 23 root root     4096 Jan 29 00:43 ..
-rw-r--r--  1 root root      118 Jan 29 00:43 armbianEnv.txt
-rw-r--r--  1 root root     1536 Jan 29 00:16 armbian_first_run.txt.template
-rw-r--r--  1 root root    38518 Jan 29 00:16 boot.bmp
-rw-r--r--  1 root root     2868 Jan 29 00:13 boot.cmd
-rw-r--r--  1 root root     4882 Jan 29 00:16 boot-desktop.png
-rw-rw-r--  1 root root     2940 Jan 29 00:43 boot.scr
-rw-r--r--  1 root root   155700 Jan 29 00:08 config-4.4.211-rockchip64-gfd449238-dirty
lrwxrwxrwx  1 root root       38 Jan 29 00:15 dtb -> dtb-4.4.211-rockchip64-gfd449238-dirty
drwxr-xr-x  3 root root     4096 Jan 29 00:15 dtb-4.4.211-rockchip64-gfd449238-dirty
lrwxrwxrwx  1 root root       42 Jan 29 00:14 Image -> vmlinux-4.4.211-rockchip64-gfd449238-dirty
-rw-r--r--  1 root root  6686333 Jan 29 00:44 initrd.img-4.4.211-gfd449238-dirty-rockchip64
-rw-r--r--  1 root root  8004169 Jan 29 00:22 initrd.img-4.4.211-rockchip64-gfd449238-dirty
-rw-r--r--  1 root root        0 Jan 29 00:14 .next
-rw-r--r--  1 root root  4844786 Jan 29 00:08 System.map-4.4.211-rockchip64-gfd449238-dirty
lrwxrwxrwx  1 root root       42 Jan 29 00:44 uInitrd -> uInitrd-4.4.211-gfd449238-dirty-rockchip64
-rw-r--r--  1 root root  6686397 Jan 29 00:44 uInitrd-4.4.211-gfd449238-dirty-rockchip64
-rw-r--r--  1 root root  8004233 Jan 29 00:22 uInitrd-4.4.211-rockchip64-gfd449238-dirty
-rwxr-xr-x  1 root root 22906888 Jan 29 00:08 vmlinux-4.4.211-rockchip64-gfd449238-dirty
```

I don't get why I get the git revision in the files. This doesn't happen for my colleague. And the final call to `update-initramfs` breaks the rootfs since it assumes the wrong kernel name and generates an initramfs without modules `uInitrd-4.4.211-gfd449238-dirty-rockchip64`.

igorpecovnik, 2020-01-29:
> When I build for the rockpro64 I get the following content in the boot directory

With a clean and un-modified build system?

NickeZ, 2020-01-29:
I build it like this:

```
./compile.sh docker BOARD=rockpro64 KERNEL_ONLY=no KERNEL_CONFIGURE=no BUILD_MINIMAL=yes BUILD_DESKTOP=no RELEASE=bionic BRANCH=legacy WIREGUARD=no PROGRESS_LOG_TO_FILE=yes
```

git status:

```
$ git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
```

igorpecovnik, 2020-01-29:
For image building under the Docker you need to enable Privileged Mode in userpatches/config-docker.conf ... I will try to recreate once I am in the office.

Also clean Docker image to force rebuilding it.

NickeZ, 2020-01-29:
I don't understand what I potentially haven't cleaned. Why do I get git repository information on my linux kernel image?

Stadicus, 2020-01-29:
I was able to recreate the issue described by @NickeZ above, here are some more details.

When generating the Armbian RockPro64 image without any userpatches (and all regular patches for the legacy branch present), I get the following `/boot` directory:

```
-rw-r--r-- 1 root root  118 Jan 29 15:15 armbianEnv.txt
-rw-r--r-- 1 root root 1.5K Jan 29 14:57 armbian_first_run.txt.template
-rw-r--r-- 1 root root  38K Jan 29 14:57 boot.bmp
-rw-r--r-- 1 root root 2.9K Jan 29 14:55 boot.cmd
-rw-r--r-- 1 root root 4.8K Jan 29 14:57 boot-desktop.png
-rw-rw-r-- 1 root root 2.9K Jan 29 15:15 boot.scr
-rw-r--r-- 1 root root 157K Jan 29 14:53 config-4.4.211-rockchip64
lrwxrwxrwx 1 root root   22 Jan 29 14:56 dtb -> dtb-4.4.211-rockchip64
drwxr-xr-x 3 root root 4.0K Jan 29 14:56 dtb-4.4.211-rockchip64
lrwxrwxrwx 1 root root   26 Jan 29 14:56 Image -> vmlinux-4.4.211-rockchip64
-rw-r--r-- 1 root root 7.5M Jan 29 15:16 initrd.img-4.4.211-rockchip64
-rw-r--r-- 1 root root 4.2M Jan 29 14:53 System.map-4.4.211-rockchip64
lrwxrwxrwx 1 root root   26 Jan 29 15:16 uInitrd -> uInitrd-4.4.211-rockchip64
-rw-r--r-- 1 root root 7.5M Jan 29 15:16 uInitrd-4.4.211-rockchip64
-rwxr-xr-x 1 root root  21M Jan 29 14:53 vmlinux-4.4.211-rockchip64
```

But when trying to build the image using the Ayufan kernel config [rockchip_linux_defconfig](https://github.com/ayufan-rock64/linux-kernel/blob/release-4.4.190/arch/arm64/configs/rockchip_linux_defconfig) copied into `userpatches/linux-rockchip64-legacy.config` as described in [this Armbian forum post](https://forum.armbian.com/topic/12861-last-effort-regarding-rockpro64-kernel-panics/?do=findComment&comment=94137), with everything else being equal I get the following directory contents:

```
-rw-r--r-- 1 root root  118 Jan 29 12:52 armbianEnv.txt
-rw-r--r-- 1 root root 1.5K Jan 29 12:34 armbian_first_run.txt.template
-rw-r--r-- 1 root root  38K Jan 29 12:34 boot.bmp
-rw-r--r-- 1 root root 2.9K Jan 29 12:33 boot.cmd
-rw-r--r-- 1 root root 4.8K Jan 29 12:34 boot-desktop.png
-rw-rw-r-- 1 root root 2.9K Jan 29 12:52 boot.scr
-rw-r--r-- 1 root root 152K Jan 29 12:28 config-4.4.211-rockchip64-gfd449238-dirty
lrwxrwxrwx 1 root root   38 Jan 29 12:34 dtb -> dtb-4.4.211-rockchip64-gfd449238-dirty
drwxr-xr-x 3 root root 4.0K Jan 29 12:34 dtb-4.4.211-rockchip64-gfd449238-dirty
lrwxrwxrwx 1 root root   42 Jan 29 12:33 Image -> vmlinux-4.4.211-rockchip64-gfd449238-dirty
-rw-r--r-- 1 root root 6.3M Jan 29 12:53 initrd.img-4.4.211-gfd449238-dirty-rockchip64
-rw-r--r-- 1 root root 7.6M Jan 29 12:42 initrd.img-4.4.211-rockchip64-gfd449238-dirty
-rw-r--r-- 1 root root 4.7M Jan 29 12:28 System.map-4.4.211-rockchip64-gfd449238-dirty
lrwxrwxrwx 1 root root   42 Jan 29 12:53 uInitrd -> uInitrd-4.4.211-gfd449238-dirty-rockchip64
-rw-r--r-- 1 root root 6.3M Jan 29 12:53 uInitrd-4.4.211-gfd449238-dirty-rockchip64
-rw-r--r-- 1 root root 7.6M Jan 29 12:42 uInitrd-4.4.211-rockchip64-gfd449238-dirty
-rwxr-xr-x 1 root root  22M Jan 29 12:28 vmlinux-4.4.211-rockchip64-gfd449238-dirty
```

igorpecovnik, 2020-01-29:
> I don't understand what I potentially haven't cleaned. 

Now I see. You used external sources which obviously needs to be patched and cleaned. This is not a build script issue, closing.

Please proceed solving this in the forum where you begun.

closed 2020-01-29 by igorpecovnik

NickeZ, 2020-01-29:
I'm sorry but I'm not a member of the forum, but yes, following the steps you have provided in the forum leads to the same issue. Could you post a solution to this issue in the forum then?

igorpecovnik, 2020-01-29:
> I'm sorry but I'm not a member of the forum

There is the only place you will receive limited support. We are understaff which means you will need to be very patient. A shortcut is do something and then ask for something.

> Why do I get git repository information on my linux kernel image?

I can assure you that issue and this question you have has nothing to do with Armbian. I double checked.

>  Could you post a solution to this issue in the forum then?

Free support doesn't include solving solutions.

NickeZ, 2020-01-29:
I've spent quite some time the last days trying to solve this issue and the only conclusion I could come to is that the line below specifies a kernel that isn't installed and therefore cannot be correct. If this project didn't author this line of code. Where can I search further assistance?

https://github.com/armbian/build/blob/1f95a0086ee9bcafacb6c00ecd7a7f6e0fd56aae/lib/debootstrap.sh#L533

I tried to the best of my ability to follow the instructions outlined by armbian here: https://docs.armbian.com/Developer-Guide_User-Configurations/#user-provided-kernel-config

After checking out the repository I just add a custom kernel config and then I build. Is there some other step I must take?

Thanks

igorpecovnik, 2020-01-30:
> I just add a custom kernel config 

Issue is your custom kernel config which you need to fix. Then things will work.

> Where can I search further assistance?

By learning Linux internals
https://training.linuxfoundation.org/training/introduction-to-linux-open-source-development-and-git/
or contracting people that would guide you.

NickeZ, 2020-01-30:
OK, I've never had this issue with any other "os builder" like yocto. Thanks for your help.

igorpecovnik, 2020-01-30:
> OK, I've never had this issue with any other "os builder" like yocto.

I have consulting contract for few hours per day for Yocto environment.

NickeZ, 2020-01-30:
Thanks for the offer.

igorpecovnik, 2020-01-30:
> Thanks for the offer.

It's not an offer - you had your chance for that. Its just a support to my claims that I am well aware of the problems of one system and another. I am co-author and a daily user of one system and advanced daily user of another. None is perfect or bug free. Each has its own advantages ...

NickeZ, 2020-01-30:
I'm not sure what offended you and how I burned any bridges but for that I'm sorry. Last few days has been my first experience ever with Armbian and honestly, this hostility you have shown me tells me that this is not a community I want to be part of. I'll use something else instead.

igorpecovnik, 2020-01-30:
> what offended you

While your colleague did an emotional blackmailing forum post, which tone I actually ignored in first place & provide you best effort answer ... while not long ago your repeated "Could you post a solution" was sadly over the limits I am willing to tolerate for support on my expense. Dealing with you makes a lot of costs and this project generates virtually no income. I simply have no desire to teach you how things are, how to work with the project, how to deal with open source projects in general in person and out of my personal pocket. Regardless of problem you have. If I am hostile to you, you will probably leave me alone and not make more damage.

That's it.

TRSx80, 2020-01-30:
For anyone reading this later, and thinking maybe too harsh reaction (which I also did at first), I wanted to leave a note that there is more to this story behind the scenes...

NickeZ, 2020-01-30:
> For anyone reading this later, and thinking maybe too harsh reaction (which I also did at first), I wanted to leave a note that there is more to this story behind the scenes...

I also want to leave a note that I was not aware of any forum post and I'm not aware of any story behind the scenes. I tried and failed to compile Armbian with a "user povided" kernel config and reached out for help here since the build system built a broken rootfs.

> I simply have no desire to teach you how things are, how to work with the project, how to deal with open source projects in general

That makes total sense to me, but since you posted I thought you were interested in helping me. The message would have been even more clear if you simply didn't post at all.

Thanks.

igorpecovnik, 2020-01-31:
> I was not aware

I believe you.

> were interested in helping me.
> even more clear

If you dare to put a pressure, our transactions becomes commercial. And we lost hours because of this.

Where to send an invoice for the service?

Stadicus, 2020-02-03:
For anybody interested, we figured out a hacky way to get `uInitrd` to link to the correct file. We changed the following line in `debootstrap.sh`:

```
-	update_initramfs_cmd="update-initramfs -uv -k ${VER}-${LINUXFAMILY}"
+	update_initramfs_cmd="update-initramfs -uv"
```

I'm not saying that's a good way to do it, and it might have unwanted side effects in different setups, but I just thought I leave that here.

https://github.com/armbian/build/blob/7084a2d457e97d99636e46f3d91a14dfe51fe8ed/lib/debootstrap.sh#L533