armbian/build #1785: [rk3399] No suspend for pcie <-> sata drives on Rockpro64

bavay, 2020-02-07:
Until the dev kernel update of 19th of January 2020, a harddrive connected to a Rockpro64 board through an ASM1062 Serial ATA Controller (on PCIe) would automatically switch to standby after the time set by hdparm. 

Since the dev kernel update of 19 Jan 2020, it never suspends... If I issue a suspend command with hdparm ('hdparm -y /dev/sda'), it does suspend though. Beside the dev kernel update, nothing changed on my system betwenn pre-01.19.2020 and post-01.19.2020

Mathias

igorpecovnik, 2020-02-07:
> Since the dev kernel update of 19 Jan 2020, it never suspends...

https://github.com/armbian/build#support

closed 2020-02-07 by igorpecovnik

bavay, 2020-02-07:
I'm sorry if I reported this bug on the wrong place, but then where should I report bugs on the dev kernel? I've mentioned this issue in the forums but I have the feeling this is not the right place either...

igorpecovnik, 2020-02-07:
> where should I report bugs on the dev kernel?

To the developers that develop and maintain areas related to the problem you are reporting? Linux kernel is a work of 1000+ people. We don't deal with the impossible - track and support Linux kernel bugs, related or unrelated to this specific hardware we deal with. Especially not in development kernels which are changing daily. Staying up2date is sadly extremely expensive and far out of a range we can pay for. You don't, so there is nothing to do here.

> I've mentioned this issue in the forums but I have the feeling this is not the right place either...

That's good enough. If issue didn't get an attention and fix, there is nothing more you can do about. Nor we can do about. It will be fixed when/if someone finds time. 

If you need this to be fixed at all costs, seek out for professionals. I would do the same if I would not be able to do it myself but would need it badly.

bavay, 2020-02-07:
No, I don't need it fixed at all costs at all. If I want something that works flawlessly, I can switch back to Ayufan's kernels (5.3.0 works perfectly for me). But somehow, I would like to use Armbian's versions in order to contribute bug reports...

So all that I want is to make sure that if somebody wants to know what works and what does not (which I guess is the role played by Armbian's dev kernel), it can be found where people would expect it. By the way, I still haven't manage to figure out where Armbian's dev kernel come from (in order to kind of build a changelog between versions that have been pushed through apt)...

igorpecovnik, 2020-02-07:
> So all that I want is to make sure that if somebody wants to know what works and what does not (which I guess is the role played by Armbian's dev kernel), it can be found where people would expect it. 

Anyone can maintain such list. It makes little value to me if I am unable to address troubles with resources there are.

> something that works flawlessly

Some board specific features could works better, agree ... but that kernel is orphan 5.3.0 without any maintenance ... we can't stay there.

Modern Rockchip current & dev is mainline + patches. 5.4.y and 5.5.y ... it will take some time when all features from Ayufan development branch are covered ... most of them are.

https://github.com/armbian/build#build-tools-overview
https://github.com/armbian/build/blob/master/config/sources/families/include/rockchip64_common.inc

https://github.com/armbian/build/tree/master/patch/kernel/rockchip64-current
https://github.com/armbian/build/tree/master/patch/kernel/rockchip64-dev

https://github.com/armbian/build/blob/master/config/kernel/linux-rockchip64-current.config
https://github.com/armbian/build/blob/master/config/kernel/linux-rockchip64-dev.config