armbian/build #1787: Errors building image for rockpi 4-b on ubuntu eoan

boxerab, 2020-02-08:
Here are the errors : 

```
[ o.k. ] Calling image customization script [ customize-image.sh ]
chroot: failed to run command ‘/bin/bash’: No such file or directory
[ error ] ERROR in function customize_image [ image-helpers.sh:117 ]
[ error ] customize-image.sh exited with error (rc: 127) 
[ o.k. ] Process terminated 
[ o.k. ] Unmounting [ /root/armbian/.tmp/rootfs-dev-rockpi-4b-buster-yes-no/ ]
[ error ] ERROR in function unmount_on_exit [ image-helpers.sh:66 ]
[ error ] debootstrap-ng was interrupted 
[ o.k. ] Process terminated 
```

igorpecovnik, 2020-02-08:
With Docker?

boxerab, 2020-02-08:
Yes:
`./compile.sh docker EXPERT="yes"`

igorpecovnik, 2020-02-08:
Try starting from scratch by removing a container:

./compile dockerpurge

I tested on Ubuntu 19.04 and it goes fine: https://pastebin.com/rctsDZn4

boxerab, 2020-02-08:
Thanks, will try that

igorpecovnik, 2020-02-08:
Also do manual Docker images removal if this doesn't help. I just found out it can fail to remove it in some cases ...

boxerab, 2020-02-08:
same error, unfortunately.  How do I do manual docker image removal ?

boxerab, 2020-02-08:
nm - figured it out.

igorpecovnik, 2020-02-11:
Not possible to recreate and apparently solved, closing.

closed 2020-02-11 by igorpecovnik