armbian/build #1798: [orangepilite/current] pcm5102a module disabled

douardda, 2020-02-11:
On a stock armbian image for the orangepi lite (image `Armbian_20.02.0-rc1_Orangepilite_buster_current_5.4.14.7z`), there is no kernel module for the pcm5102 (i2s with no hw control, so no i2c/spi required) DAC.

Trying to build a kernel using the build toolchain, I noticed for some reason, the pcm5102a build sound module is not even listed in the modules (in the allwinner soc codecs submenu) while configuring a kernel build because of the lack of a description for the entry `config SND_SOC_PCM5102A` in the `linux-mainline/orange-pi-5.4/sound/soc/codecs/Kconfig` file. Replacing:
```
config SND_SOC_PCM5102A
        tristate
```
with 
```
config SND_SOC_PCM5102A
        tristate "Texas Instruments PCM5102A"
```

makes it possible to build the module (and it work!), with the help of an overlay like:

```
/dts-v1/;
/plugin/;

/{
  compatible = "allwinner,sun8i-h3";

  fragment@0 {
    target-path = "/";
    __overlay__ {
      pcm5102a: pcm5102a {
        #sound-dai-cells = <0>;
        compatible = "ti,pcm5102a";
        pcm510x,format = "i2s";
      };
    };
  };

  fragment@1 {
    target = <&i2s0>;
    __overlay__ {
      //compatible = "allwinner,sun8i-h3-i2s";
      status = "okay";
      pinctrl-0 = <&i2s0_pins>;
      sound-dai = <&pcm5102a>;
      //clock-source = "external";
      pinctrl-names = "default";
    };
  };

  fragment@2 {
    target-path = "/";
    __overlay__ {
      sound_i2s {
        compatible = "simple-audio-card";
        simple-audio-card,name = "I2S-master";
        simple-audio-card,mclk-fs = <256>;
        simple-audio-card,format = "i2s";
        status = "okay";
        simple-audio-card,cpu {
          sound-dai = <&i2s0>;
        };
        simple-audio-card,codec {
          sound-dai = <&pcm5102a>;
        };
      };
    };
  };
};
```

I'd be nice to have it already built in armbian to make life easier for those who want to play with these cheapo but very capable DACs. 

I'm not sure where to start if I want to provide a PR for this. I guess I have to provide a patch in `patch/kernel/sunxi-current` right?

lanefu, 2020-02-12:
@igorpecovnik @martinayotte any thoughts?

igorpecovnik, 2020-02-12:
> I guess I have to provide a patch in patch/kernel/sunxi-current right?

If its tested, yes, here. And to patch/kernel/sunxi-dev , optional sunxi-legacy

ashthespy, 2020-02-12:
^If its tested, yes, here. And to patch/kernel/sunxi-dev , optional sunxi-legacy

Been using the PCM5102a with the orangepione/lite using a similar method with the legacy (4.19.y) kernel. It did take some head scratching initially wondering why the module wasn't getting compiled directly from Kconfig then came across [this forum thread](https://forum.armbian.com/topic/9009-info-friendlyarm-pcm5102a-hat-with-nanopi-neo-under-mainline-4xx-and-dev-533/) that suggested the same description fix. 

```diff 
diff --git a/sound/soc/sunxi/Kconfig b/sound/soc/sunxi/Kconfig
index 22408bc..73126bd 100644
--- a/sound/soc/sunxi/Kconfig
+++ b/sound/soc/sunxi/Kconfig
@@ -30,8 +30,10 @@ config SND_SUN8I_CODEC_ANALOG
 
 config SND_SUN4I_I2S
 	tristate "Allwinner A10 I2S Support"
+	depends on OF
 	select SND_SOC_GENERIC_DMAENGINE_PCM
 	select REGMAP_MMIO
+	select SND_SOC_PCM5102A
 	help
 	  Say Y or M if you want to add support for codecs attached to
 	  the Allwinner A10 I2S. You will also need to select the
```

PR #1802 (2020-02-14): Allow pcm5102 module to be built on sunxi kernels and build the module by default
This PR consist in 2 commits: 
- the first one makes it possible to select the PCM5102A module when configuring the build of sunxi-current and sunxi-dev kernels,
- the second one makes pcm5102a as well as pcm512x_(sdi|i2c) modules built by default, since these are very commonly found on cheap DAC breakboards or hats.

This sould fix #1798
commit 76dc8aba (2020-02-14): Allow pcm5102 module to be built on sunxi kernels and build the module by default (#1802)

* [sunxi] allow pcm5102a codec module to be built

for sunxi-current and sunx-dev kernels.

Fixes #1798

* [sunxi] build pcm5102a and pcm51xx_(i2s|spi) modules by default

on sunxi-current and sunxi-dev kernels.
closed 2020-02-14 by igorpecovnik