armbian/build #1812: [security] Kr00k

ThomasKaiser, 2020-02-26:
Since most if not all Armbian supported wireless capable boards use Broadcom/Cypress chipsets and the RPi 3 tested positive I guess all these devices are affected by [Kr00k](https://www.eset.com/int/kr00k/)?

EvilOlaf, 2020-02-27:
Hard to find any useful information around. Some threads popped up at Pihole and raspberry pi forums but nothing really yet.
https://nvd.nist.gov/vuln/detail/CVE-2019-15126
Seem to sit mainly in the bcm firmware which has to be patched by the manufacturer.
Maybe more affected chips will be named over time to be affected as well. For example the OPi Zero Plus H5 comes as per the data sheet with Realtek RTL8189FTV as WiFi.

ThomasKaiser, 2020-02-27:
> Hard to find any useful information around.

Huh? That we've heard about this publicly just yesterday is due to their rather responsible vulnerability disclosure policy. But the issue is well documented and the fact that the stuff [here](https://github.com/armbian/firmware) is rather outdated means at least the BCM equipped boards suffer from insecure wireless operation that is known to everyone since yesterday.

ThomasKaiser, 2020-02-27:
Take this for example: [Igor get's some files](https://github.com/armbian/firmware/commit/9f18a83f618faff86cf368cf9589ba98653a7667) most probably from @hipboi and throws it in a repo where the giant armbian-firmware package is generated from. And from the on it's rotting forever there and on user's devices.

Broadcom fixed the stuff within the last months, they notified their 'customers' to provide this stuff as patches. So in an ideal world those vendors would now also fix their stuff (provide new firmware BLOBs) and push this out to users or 3rd parties like Armbian. But I guess the SBC world is not an ideal world...

EvilOlaf, 2020-02-27:
...and probably never will.

I tried to find something even though I am not very deep into this topic. What I could find was The firmware-brcm80211 Debian package but this seem not to have received fixes yet. I also could not find (yet) any new upstream firmware blobs, maybe they are non-public? Would not make much sense though...

ThomasKaiser, 2020-02-27:
See also https://forum.armbian.com/topic/4949-security-broadpwn/ (Armbian and wireless security is essentially a non-issue since nobody gives a sh*t). Back then when I was really dumb and maintained some OMV ARM images I took care to replace the `armbian-firmware` package in favor of Raspbian's `firmware-nonfree` to fix this stuff at least on the most popular OMV for ARM image.

EvilOlaf, 2020-02-27:
Sometimes you are on your own to make the world a bit better. This makes it not less valuable. In my eyes at least.

EvilOlaf, 2020-02-27:
The Broadcom blobs in either the Xulong repo and the Debian package seem to be quite outdated. `strings` from each binary revealed dates from 2011 to 2018, nothing newer.

igorpecovnik, 2020-02-27:
Can we fully ditch our firmware packages for upstream?https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git ... 

Who will check that?

EvilOlaf, 2020-02-27:
This would not fix the security flaw since those blobs seem to be quite old as well. The commit log confirms my results from `strings`: 
https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/log/brcm

igorpecovnik, 2020-02-27:
> This would not fix the security flaw since those blobs seem to be quite old as well.

So this will again require to provide better services as corporation supported upstream https://www.linuxfoundation.org

ThomasKaiser, 2020-02-27:
The Debian packages `firmware-b43legacy-installer_019-4_all` and `firmware-b43-installer_019-4_all` contain no actual data but do it this way via their `postinst` script:

The first one sucks random stuff from the Internet:

    VERSION=""	
    BROADCOM_WL=""	
    WL_APSTA="wl_apsta-3.130.20.0.o"	
    DOWNLOAD="${WL_APSTA}"	
    URL="http://downloads.openwrt.org/sources/${WL_APSTA}"	
    SHA512SUM="d89ed52045307449bbae79a4d1807cc6cd89ae67c4a22e8e8aa51c1396edbb6ed8b157cd0756faf8b660a537b48b62117c57967f2048245b5b102d9d9bca4bbd"	
    FIRMWARE_INSTALL_DIR="/lib/firmware"	
    B43="b43legacy"	

And the 2nd does the same:

    VERSION="5.100.138"	
    BROADCOM_WL="broadcom-wl-${VERSION}"	
    WL_APSTA="${BROADCOM_WL}/linux/wl_apsta.o"	
    DOWNLOAD="${BROADCOM_WL}.tar.bz2"	
    URL="http://www.lwfinger.com/b43-firmware/${DOWNLOAD}"	
    SHA512SUM="02487e76e3eca7fe97ce2ad7dc9c5d39fac82b8d5f7786cce047f9c85e2426f5b7ea085d84c7d4aae43e0fe348d603e3229211bab601726794ef633441d37a8b"	
    FIRMWARE_INSTALL_DIR="/lib/firmware"	
    B43="b43"

hipboi, 2020-02-28:
We already contacted Ampak for the new firmware. Will update here once we get the reply.

ThomasKaiser, 2020-02-28:
> Will update here once we get the reply.

Thank you Tom! Hope to hear from you soon.

@igorpecovnik in my opinion this flaw could serve as some sort of a testimonial which board makers do care about security and which don't. IMO it should be mentioned on the download page in a way such as 'Wi-Fi vulnerable to [BroadPwn](https://nvd.nist.gov/vuln/detail/CVE-2017-9417) and [Kr00k](https://nvd.nist.gov/vuln/detail/CVE-2019-15126)' (maybe even in big red letters) vs. 'Common Wi-Fi vulnerabilities like [BroadPwn](https://nvd.nist.gov/vuln/detail/CVE-2017-9417) and [Kr00k](https://nvd.nist.gov/vuln/detail/CVE-2019-15126) fixed'.

EvilOlaf, 2020-02-28:
>  'Wi-Fi vulnerable to [BroadPwn](https://nvd.nist.gov/vuln/detail/CVE-2017-9417) and [Kr00k](https://nvd.nist.gov/vuln/detail/CVE-2019-15126)'...
...and there is no way for Armbian to fix this without the help of the chipset manufacturer, so please blame them.

😄

ThomasKaiser, 2020-02-28:
> chipset manufacturer

It's the **board manufacturer's** duty to get in contact with Ampak, pull a new firmware from them, test it with their devices and release it to the public. The fixes exist already and have been pushed to majority of vulnerable devices within the last months.

It's just that almost nobody in the SBC world seems to care about wireless (in)security and as such we're dealing with some smelly BLOBs from ages ago that whoever found on random places on the Internet and this gets bundled as `armbian-firmware` package instead of being called `random-old-junk-found-here-and-there` package to illustrate the process behind.

If Armbian wants to improve on this situation now is the time to taunt board makers...

hipboi, 2020-03-02:
Next week we will get the new updated/fixed firmware.

EvilOlaf, 2020-03-05:
There is some communication with Cypress going on I think:
https://github.com/linux-mailinglist-archives/linux-kernel.vger.kernel.org.0/blob/15e842ac08ee319a3e941a0b67f8acb8d77fee6e/m

igorpecovnik, 2020-03-05:
[1] [Radxa](https://github.com/armbian/firmware/commit/2a7e18f05305277c9fb22443c6343ad3fa1d2f5d) merged and [pushed](https://github.com/armbian/upload/commit/7bf5d7aeb980bc55b3e8fd9f28521f6d0f9a0425)

ThomasKaiser, 2020-03-05:
What about the other board makers? Tom/Radxa delivered amazingly fast which is great. :)

This Github issue makes it rather easy to point other board makers to and blame them if they don't react within n days time. Why not doing this? Mentioning them like @wuweidong0107 for example is rather easy.

igorpecovnik, 2020-03-05:
> Why not doing this?

@leeboby @dangku Please help us sorting out this issue! Thanks.

sfx2000, 2020-03-09:
> The Debian packages `firmware-b43legacy-installer_019-4_all` and `firmware-b43-installer_019-4_all` contain no actual data but do it this way via their `postinst` script:

This is all legacy stuff - broadcom softmac/fullmac is beyond this with newer chipsets.

Still blobs - just saying... some of this is going to be from Cypress, some from Broadcom.

saurav003e, 2020-03-12:
is the code exposed in internet ??

EvilOlaf, 2020-03-20:
Now there is...

ThomasKaiser, 2020-04-02:
So to summarize 5 weeks later. Except of [Radxa/RockPi](https://github.com/armbian/firmware/commit/2a7e18f05305277c9fb22443c6343ad3fa1d2f5d) and the [RPi Trading guys](https://github.com/RPi-Distro/firmware-nonfree/commit/ce751a84617715ca0189e55c3cf07f12d87e16e5) the rest of the ARM world simply doesn't give a shit about broken wireless security.

igorpecovnik, 2020-04-02:
> world simply doesn't give a shit about broken wireless security.

... but they waste resources to provide a bad version of what we do, but its their official, or do nothing but sale stuff.

saurav003e, 2020-04-02:
Ok bro ....I got it....I knew it ....its a big wifi vulnerability....and
when I search I saw your code....then I think how a big vulnerability is
realised soon .....

I commended because I saw this code and you give how to run that code....so
when i run its seems error....that's why I commended