armbian/build #1818: [new feature] Banana PI R64 support

muravjov, 2020-02-28:
Hi,

I would like to try to (help to) port what I know about R64 into code in armbian/build to generate a buster image.
I see that R2 is already ported to Armbian, and the kernel with Frank's improvements is used,
https://github.com/armbian/build/blob/master/config/sources/families/mt7623.conf#L15

To start, I built R2's image and began investigate how it works and got a question about booting the kernel - armbian seems to use these 2 scripts to boot it: `/boot/armbianEnv.txt`, `/boot/boot.cmd`.
I don't see how to change kernel in them?

I got used to change that in /boot/bananapi/bpi-r64/linux/uEnv.txt, providing both kernels are already installed, like so:
```
#kernel=uImage_nodt_old
#fdt=bpi-r64_old.dtb

kernel=uImage_nodt
fdt=bpi-r64.dtb
```

My `/boot/armbianEnv.txt`, `/boot/boot.cmd`
```
# cat /boot/armbianEnv.txt 
verbosity=1
rootdev=UUID=a5d0bb56-3b50-41fe-9c0b-fc20928127b7
rootfstype=ext4

# ls /boot/boot.cmd 
/boot/boot.cmd
root@ilya-thinkpad:/# cat /boot/boot.cmd   
# DO NOT EDIT THIS FILE
#
# Please edit /boot/armbianEnv.txt to set supported parameters
#
setenv rootdev "/dev/mmcblk1p1"
setenv rootfs "ext4"
setenv verbosity "1"
setenv devtype "mmc"
setenv prefix "boot/"

echo "Boot script loaded from device ${devnum}"

if test -e ${devtype} ${devnum} ${prefix}armbianEnv.txt; then
        load ${devtype} ${devnum} ${kernel_addr_r} ${prefix}armbianEnv.txt
        env import -t ${kernel_addr_r} ${filesize}
fi

setenv bootargs "console=ttyS2,115200n1 root=${rootdev} rw rootfstype=${rootfs} rootwait audit=0 loglevel=${verbosity}"
ext4load ${devtype} ${devnum}:${mmcpart} ${fdtaddr} ${mmcfdtfile}
ext4load ${devtype} ${devnum}:${mmcpart} ${rdaddr} ${mmcinitrdfile}
ext4load ${devtype} ${devnum}:${mmcpart} ${kernel_addr_r} ${mmckernfile}
echo "Booting ${mmckernfile} ${mmcinitrdfile} ${mmcfdtfile} from: ${devtype} ${devnum}:${mmcpart} using bootargs=${bootargs}"
bootz ${kernel_addr_r} ${rdaddr} ${fdtaddr}
# Recompile with:
# mkimage -C none -A arm -T script -d /boot/boot.cmd /boot/boot.scr

```

martinayotte, 2020-02-28:
> I don't see how to change kernel in them?

You don't need to change any things in /boot/boot.cmd or /boot/armbianEnv.txt to switch across different kernels, you just need to switch the symbolic links located in /boot/ ...

muravjov, 2020-02-28:
These two links?
```
# ls -lah /boot/ | grep -E '(zImage|dtb)'
lrwxrwxrwx  1 root root   17 Feb 28 00:41 dtb -> dtb-5.4.22-mt7623
drwxr-xr-x  2 root root 4.0K Feb 28 00:41 dtb-5.4.22-mt7623
lrwxrwxrwx  1 root root   21 Feb 28 00:38 zImage -> vmlinuz-5.4.22-mt7623
```
I need to change them by hand or exists some utility to change that? Is there documentation about it?

martinayotte, 2020-02-28:
Yes, but also a third one : /boot/uInitrd 

> I need to change them by hand

If you copied kernel files manually, including /lib/modules/<kernel_modules_version>, yes, links need to be switch by hand. Otherwise, if installed by doing "dpkg -i <armbian_image>.deb", it should take care of the links automatically ...

chwe17, 2020-03-02:
I don't think starting from a w2 image is a good idea. First MT7622 is 64bit whereas mt7623n was 32bit.. I've no idea about upstream u-boot support (whereas this works well with mt7623n). And you likely want a 64bit kernel combined with a 64bit rootfs on this board. Just installing a mt7622 trimmed kernel over a R2 image likely won't work. On the other side, we use some tweaks to get the network up which are likely not needed for the mt7622. Starting from plain might be the better option.

muravjov, 2020-03-03:
> I don't think starting from a w2 image is a good idea. First MT7622 is 64bit whereas mt7623n was 32bit.. I've no idea about upstream u-boot support (whereas this works well with mt7623n). 

Yes, I mean that I needed an example to learn from (banana r2 is fine for that). I do work in https://github.com/muravjov/armbian-build and received a working system, e.g. one can try it https://github.com/muravjov/armbian-build/releases/tag/2020-03-02_c539c65

Status: 
* userspace is ready
* kernel is used from https://github.com/frank-w/BPI-R2-4.14
* u-boot: is prebuilt, also from https://github.com/frank-w

I am not so good at u-boot so it needs time to build it in Armbian way and make use of /boot/armbianEnv.txt instead of /boot/bananapi/bpi-r64/linux/uEnv.txt

igorpecovnik, 2020-12-17:
How are things going at R2? Can we move it in with 5.10.y?

martinayotte, 2020-12-17:
Hi Igor !
I don't have any R2/R64 boards ... 
DEV is already on 5.10.y, do you mean that you wish to switch CURRENT ?

igorpecovnik, 2020-12-17:
Yeah, I mean moving to 5.10.y or which is most relevant ... We only have some old instance https://github.com/armbian/build/blob/master/config/sources/families/mt7623.conf#L10-L14

Perhaps op @muravjov know something about? Is it mainline good enough? I see there is some movement at @frank-w repository but I don't pay attention to this hw. Even I have them - somewhere.

muravjov, 2020-12-17:
> Perhaps op @muravjov know something about? Is it mainline good enough? I see there is some movement at @frank-w repository but I don't pay attention to this hw. Even I have them - somewhere.

`linux-4.19` is maybe still relevant for R2, it is definitely irrelevant for R64 (i am interested in R64 only).

igorpecovnik, 2020-12-17:
> i am interested in R64 only

Got it. Are you still interested to push R64 to Armbian? This way can also reach out to more people.

muravjov, 2020-12-17:
All my development with R64 is frozen now, I would like continue next year likely, but cannot promise

igorpecovnik, 2022-07-31:
Closing, stalled. Feel free to open if things changes.

closed 2022-07-31 by igorpecovnik