armbian/build #1824: Can I use the srcipts on arm

ChenYifei21, 2020-03-02:
I want build Armbian on my orangepi3. Because I don't hava any computer. But it says "Only on x64,arm it's unsupported.
Who can help me？I only want use custom uboot and kernel on my board. If you kown how can use the custom kernel on Armbian. Please send an email to 13997824757@139.com
Thanks!

zhangn1985, 2020-03-02:
arm board is too weak to build. and the thermal condition is bad in compare of x86 pc.

igorpecovnik, 2020-03-02:
There are more reasons like complexity and cost. Other build tools also does not support that:
https://github.com/armbian/build#compare-with-industry-standards

That doesn't mean its impossible ...

> Please send an email to 13997824757@139.com

Personalised support?
https://github.com/armbian/build#support

closed 2020-03-02 by igorpecovnik