armbian/build #1826: Armhf userspace support

Askmewho, 2020-03-02:
Hi! I am testing panfrost and lima drivers on armbian, but my main problem it's the userspace. While chroots are an option for testing.. it's not that userfriendly to other people to run inside them. Also, multiarch it's very problematic for both Mesa (they goes in conflict). I beg you to add the chance to build armhf userspace on the script.

igorpecovnik, 2020-03-02:
> build armhf userspace on the script.

We considered that [years ago](https://github.com/armbian/build/issues/645), but for other reasons. Project that generates no income can't cover [wishes which are already several years / million dollars long](https://forum.armbian.com/forum/38-feature-requests/). 

I [beg you](https://www.armbian.com/donate/) to help every day to be able to hire full time staff one day ... but we are also years away.

Askmewho, 2020-03-02:
Haha begging it's free so I will continue begging. Armhf keeps being the standard, even nowdays.. so, armbian should support standard, no just arm64. You are the leader and it's your choice, obviusly... Best regards

igorpecovnik, 2020-03-02:
We support armhf on 32bit ARM HW. Just not on arm64. That one has only native arm64 support.

> and it's your choice

This is community driven development. If you need / want / wish something, make it happen. https://www.armbian.com/get-involved/

martinayotte, 2020-03-03:
> Armhf keeps being the standard, even nowdays..

No, Native is the standard. If you wish to get armhf on arm64, you need to use multiarch.
BTW, on x86-64 machine, did you ever saw some x86-32 kernel available ?

Askmewho, 2020-03-03:
I am not asking for 32 bit kernel. Also, multiarch seems to work better on x86 than on arm.at least on my experiences.

xlazom00, 2020-04-18:
@Askmewho try this ppa it is for x86 x64 and arm64
https://launchpad.net/~oibaf/+archive/ubuntu/graphics-drivers
and for armhf you can rebuild it from source package. it is really easy

closed 2020-07-01 by lanefu