armbian/build #1832: Error building - recipe for target 'vmlinux' failed

jolla, 2020-03-03:
Build fails when attempting to build an image for nanopim4v2 if "Show a kernel configuration menu before compilation" is selected. 

Makefile:1077: recipe for target 'vmlinux' failed
[ error ] ERROR in function compile_kernel [ compilation.sh:378 ]
[ error ] Kernel was not built [ @host ]
[ o.k. ] Process terminated

EvilOlaf, 2020-03-04:
Check your `output/debug` folder for further investigation.

igorpecovnik, 2020-03-04:
I was not able to recreate. Both legacy 4.4.y and current 5.4.y kernels builds without issues, DEV kernels are not relevant.

In case you are changing a kernel configuration and enable a feature that is broken / not compatible, there is little we can do about.

jolla, 2020-03-04:
I was trying to added iwlwifi drivers, after receiving the mentioned error I repeated the same process but only selected the option for "Show a kernel configuration menu before compilation" but didn't make any changes to the kernel configuration, I still received the same error. The reason I suspected it is related to the kernel configuration option is that the builder compiles the img successfully as long as I don't select the option to customize the kernel configuration.

Perhaps this provides some clue?

wlanpi@wlanpi-virtual-machine:~/armbian/build/output/debug$ tail compilation.log 
/home/wlanpi/armbian/build/cache/sources/linux-mainline/linux-5.4.y/./include/net/cfg80211.h:6146: undefined reference to `__cfg80211_alloc_reply_skb'
/home/wlanpi/armbian/build/cache/toolchains/gcc-arm-8.3-2019.03-x86_64-aarch64-linux-gnu/bin/aarch64-linux-gnu-ld: drivers/net/wireless/rtl8812au/os_dep/linux/rtw_cfgvendor.o: in function `rtw_cfgvendor_send_cmd_reply':
/home/wlanpi/armbian/build/cache/sources/linux-mainline/linux-5.4.y/drivers/net/wireless/rtl8812au/os_dep/linux/rtw_cfgvendor.c:209: undefined reference to `cfg80211_vendor_cmd_reply'
/home/wlanpi/armbian/build/cache/toolchains/gcc-arm-8.3-2019.03-x86_64-aarch64-linux-gnu/bin/aarch64-linux-gnu-ld: drivers/net/wireless/rtl8812au/os_dep/linux/rtw_cfgvendor.o: in function `cfg80211_vendor_cmd_alloc_reply_skb':
/home/wlanpi/armbian/build/cache/sources/linux-mainline/linux-5.4.y/./include/net/cfg80211.h:6146: undefined reference to `__cfg80211_alloc_reply_skb'
/home/wlanpi/armbian/build/cache/toolchains/gcc-arm-8.3-2019.03-x86_64-aarch64-linux-gnu/bin/aarch64-linux-gnu-ld: drivers/net/wireless/rtl8812au/os_dep/linux/rtw_cfgvendor.o: in function `rtw_cfgvendor_get_feature_set_matrix':
/home/wlanpi/armbian/build/cache/sources/linux-mainline/linux-5.4.y/drivers/net/wireless/rtl8812au/os_dep/linux/rtw_cfgvendor.c:383: undefined reference to `cfg80211_vendor_cmd_reply'
/home/wlanpi/armbian/build/cache/toolchains/gcc-arm-8.3-2019.03-x86_64-aarch64-linux-gnu/bin/aarch64-linux-gnu-ld: drivers/net/wireless/rtl8812au/os_dep/linux/wifi_regd.o: in function `_rtw_regd_init_wiphy':
/home/wlanpi/armbian/build/cache/sources/linux-mainline/linux-5.4.y/drivers/net/wireless/rtl8812au/os_dep/linux/wifi_regd.c:395: undefined reference to `wiphy_apply_custom_regulatory'
make: *** [vmlinux] Error 1

jolla, 2020-03-04:
Ok, based on the errors in compilation.log I was able to find the culprit, I was selecting Y when I received this prompt.

[image]

I re-ran the builder with selecting N for 88XXAU and it was able to compile the IMG successfully.

Thanks for the help!

closed 2020-03-04 by jolla

igorpecovnik, 2020-03-04:
AFAIK selecting "M" should also be good.