armbian/build #1837: [discussion] split armbian firmware package

zhangn1985, 2020-03-10:
most of firmwares are required by components soldered on boards or a part of SOC.
each board/family knows which firmwares are required.

and legacy kernel may requires different firmwares, broadcom chips as example, dhd driver is used for legacy kernel, it requires ap_xxxxx, and brcmfmac is used for mainline kernel, it requires brcmfmac_xxx.

based on this two reasons, split armbian firmware package would be a good next step.

at armbian build library level:

build_firmware_package(){
    prepare playground 
    family_firmware_install()
    board_firmware_install()
    build board firmware
}

finally armbian-firmware-BOARD-BRANCH_VERSION-ARCH.deb will be created.

family_firmware_install is interface for SBC family
board_firmware_install is interface for board.

zhangn1985, 2020-03-10:
a new branch is created with my new patch.

https://github.com/armbian/build/tree/firmware

igorpecovnik, 2020-03-25:
OK. We got firmware package and install logic per family and branch + override per board.

How to organise fw files to make this work?

zhangn1985, 2020-03-26:
I don't fully understand the question.
in my plan, 
just put them in current armbian-firmware-git, if required firmwares are not in upstream linux-firmware git. borad/family maintainers pick them from these two git, re-organize in the way kernel wants.

and firmware packages/debs service kernel driver, so they only relate to the kernel version: branch,  family and revision. no relation to distro, debian or ubuntu.

zhangn1985, 2020-04-01:
Hi, maintainers:
here is an example for amlogic's vdec firmware,
https://github.com/armbian/build/commit/2b2d368ffe1c6f3ec5f381caf111de7cffe981f4

igorpecovnik, 2020-04-01:
Looks fine for me.

Now how to deal with exceptions and border cases - when one wants to deploy full armbian-firmware, where boards wifi  might broke, but his 3rd party hardware will receive proper fw.

Perhaps we can leave those questions for a meeting?

zhangn1985, 2020-04-01:
yes, these questions are not taken into account in the first place.

zhangn1985, 2020-04-01:
maybe we can have a armbian-firmware as a meta-package, it **suggests** to install family-firmware package, and armbian-firmware-common package.
armbian-firmware-common is a pacakge provides firmware for armbian supported, but family independent HW, eg external wifi/bt chip.

closed 2023-12-20 by zhangn1985